steve-community/steve #252: stopTransaction after tag expiry

eviontech, 2019-12-16:
If a ocpp tag starts a transaction and then while transaction is active it expires

currently there is no way to stop the transaction

goekay, 2019-12-16:
> currently there is no way to stop the transaction

can you elaborate on that? the transaction is stopped by the station.

eviontech, 2019-12-16:
a transaction was started by station using a ocpp tag

while transaction was not completed the tag expired

now station cannot stop the transaction

goekay, 2019-12-16:
> now station cannot stop the transaction

this is what i wrote, as well. the station decides. not steve. the backend cannot do anything about this. maybe you should raise this problem by the station manufacturer.

eviontech, 2019-12-16:
let me check if station can stop the trannsaction

goekay, 2019-12-16:
i want to remind you once again to follow the guidelines of the issue template. by not following the guidelines you do not give the full context in detail for me to help you. you are opening issues that are about problems unrelated to steve, as well.

closed 2019-12-16 by goekay