steve-community/steve #254: stopTransaction on other tag

eviontech, 2019-12-17:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.1
Operating system  : ...
JDK               : ...
Database          : 1.0.2
```

#### Expected Behavior
When we start a transaction with a tag the stop transaction should happen with same tag
the stop transaction method with any other tag should not be accepted
...

#### Actual Behavior
the transaction was started with  a tag and when stop transaction request was sent from other tag it was accepted
...

#### Steps to Reproduce the Problem

  1. start a transaction with 1 tag
  2. end transaction with other tag (Not related/ not parent tag)
  3. ...

#### Additional context

...
[image]

goekay, 2019-12-17:
StartTransaction and StopTransaction messages _are not requests_. they are just _notifications_ about an already started or stopped transaction. the station decides. therefore, the backend (steve) cannot react to a StopTransaction as you wish it did. the idtag in a StopTransaction is just there to get an up-to-date authorization status from the backend for the station to update its local authorization cache. there isn't even a fitting status value to represent your situation.

it looks like a charging station problem. as a general note, you can look into the ocpp specification to better understand the protocol, behaviour and use cases.

eviontech, 2019-12-17:
i am not pro at modifying the application

it would be a great help if you can guide me on implementing the feature on the installation of rejecting stop transaction if the tag is not the same as the starting tag (Parents tag should be allowed)

eviontech, 2019-12-17:
[image]

Can it be done this way ? also i need help on how to send rejected response in else part

csamsel, 2019-12-17:
As mentioned previously StopTransaction is a notification it cannot be rejected. 

You need to change the charging station.

closed 2019-12-17 by goekay