steve-community/steve #259: Submitting Update Firmware.conf results in “Error deserializing the payload of the incoming response”

DeltaVetal26, 2019-12-21:
Hi,

I`m creating UpdateFirmware function for ChargePoint.

What's happening:
1) Steve send to CP update firmare message (UpdateFirmware.req):

> [2,"a5a0ffa5-1804-4225-ab36-27a1ecd53f62","UpdateFirmware",{"retrieveDate":"2019-12-22T00:00:00.000Z","location":"https://google.com","retries":3,"retryInterval":5}]

2) CP success getted message.

3) CP sended to Steve respond (UpdateFirmware.conf)

> [3,"a5a0ffa5-1804-4225-ab36-27a1ecd53f62",{"status":"Accepted"}]

4) Steve closing connection with code:

> de.rwth.idsg.steve.SteveException: Deserialization of incoming response payload failed

> de.rwth.idsg.steve.SteveException: Deserialization of incoming response payload failed
	at de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer.handleResult(Deserializer.java:162) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer.accept(Deserializer.java:82) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.IncomingPipeline.accept(IncomingPipeline.java:49) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleTextMessage(AbstractWebSocketEndpoint.java:108) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.onMessage(AbstractWebSocketEndpoint.java:80) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ConcurrentWebSocketHandler.handleMessage(ConcurrentWebSocketHandler.java:50) ~[steve.jar:?]
	at org.springframework.web.socket.adapter.jetty.JettyWebSocketHandlerAdapter.onWebSocketText(JettyWebSocketHandlerAdapter.java:83) ~[spring-websocket-5.2.1.RELEASE.jar:5.2.1.RELEASE]
	at jdk.internal.reflect.GeneratedMethodAccessor63.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:71) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:241) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:65) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:233) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:150) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:319) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:201) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:226) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.Parser.parseSingleFrame(Parser.java:260) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.readParse(AbstractWebSocketConnection.java:583) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:441) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:428) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:781) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:917) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at java.lang.Thread.run(Thread.java:834) [?:?]
Caused by: com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "status" (class ocpp.cp._2015._10.UpdateFirmwareResponse), not marked as ignorable (0 known properties: ])
 at [Source: UNKNOWN; line: -1, column: -1] (through reference chain: ocpp.cp._2015._10.UpdateFirmwareResponse["status"])
	at com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException.from(UnrecognizedPropertyException.java:61) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownProperty(DeserializationContext.java:840) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.deser.std.StdDeserializer.handleUnknownProperty(StdDeserializer.java:1192) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownProperty(BeanDeserializerBase.java:1592) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownVanilla(BeanDeserializerBase.java:1570) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:294) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:151) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.ObjectMapper._readValue(ObjectMapper.java:4173) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2467) ~[jackson-databind-2.10.1.jar:2.10.1]
	at com.fasterxml.jackson.databind.ObjectMapper.treeToValue(ObjectMapper.java:2920) ~[jackson-databind-2.10.1.jar:2.10.1]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.Deserializer.handleResult(Deserializer.java:160) ~[steve.jar:?]
	... 34 more

Most likely the reason is the incorrect CP response to the update request from Steve. In the OCPP 1.6 documentation, under "6.56. UpdateFirmware.conf"

There is nothing about the structure of the answer. I decided to send just status: Accepted.

DeltaVetal26, 2019-12-21:
This problem is resolved by sending an empty payload.

> [3,"75d559b5-f4fb-4bbb-bd42-ffd301c278e7",{}]

closed 2019-12-21 by DeltaVetal26

goekay, 2019-12-21:
yes, because the spec defines no fields, which means the response message should be empty, which translates to `{}`.