steve-community/steve #265: need a install description

Jeepersfinest, 2020-01-08:
#### Checklist

- [X ] I checked other issues already and found no answer
- [X ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve
Hi, i tried all guides to install SteVe, but running allways in a different bad situation.
For example:
Ubuntu 18.04 in Docker (on Qnap)
Ubuntu in VM on MacOS
Ubuntu with Docker on MacOS
with mySQL 5.7 and Maria DB 10.4
I´m not an expert within Linux and need a more detailed description.
Is their an updated guide to install Steve?

Thanks in advanced
Christian
...

#### Describe the solution you'd like

...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

goekay, 2020-01-08:
hi, actually the guide is still up-to-date. what are the problems you encounter?

Jeepersfinest, 2020-01-11:
Hi, i tried a lot of different installations.
I use a QNAP-NAS with actual fimware and installed in "Container station" the official Image Ubuntu 18.04 LTS as a docker. So far, so good.
Because of the restrictions in the docker, i trieded to use QNAP-onboard MYSQL as SQL-Server.
Installations works (=DB-Connection) until the break because of MySQL 5.5.
There is no way to upgrade the Onboard-SQL, so i installed MariaDB 10.4.7 in parallel.  
The only differnet between the databases is the port, 3308 is MariaDB, 3306 MySQL 5.5.
Rest is equal, timezone, rights on db, etc...
I changed the port in Main.Properties and run "mvn package":

> [ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:5.2.4:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException:                                                      
[ERROR] Unable to obtain connection from database (jdbc:mysql://192.168.100.72:3308/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure                                               
[ERROR]                                                                                                                                                                                                                 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.                                                                                
[ERROR]                                                                                                                                  
[ERROR] SQL State  : 08S01                                                                                                                                                                                              
[ERROR] Error Code : 0                                                                                                                                                                                                  
[ERROR] Message    : Communications link failure   <

:-(
Any ideas?

Thanks in advanced!
Christian

csamsel, 2020-01-11:
Ubuntu inside docker on a NAS with external MySQL sounds all very wierd. There is a lot that can go wrong i guess. Your log file shows that the database server is not reachable.

If you dont feel you have the knowledge to make this setup please use a vanilla Ubuntu VM or use an Ubuntu installation on Raspberry Pi. Which that we can probably help you as this is easy for us to reproduce.

Jeepersfinest, 2020-01-14:
Okay. Thanks a lot!

closed 2020-01-14 by Jeepersfinest