steve-community/steve #270: Add "Unknown Charge Point": ChargePoint ID is altered

ue-lee, 2020-01-14:
#### Checklist

#### Specifications
```
SteVe Version     :  3.4.1
Operating system  : Debian 10
JDK               : 11.0.5
Database          : MySQL 14.14, Distribution 5.7.29
```

#### Expected Behavior
Webinterface: Data Management -> Charge Points
After pushing "Add" in "Unknown Charge Points" for e.g. ChargeBox ID "15999999_19999.0B", Charge Point should vanish from this category ("Unknown Charge Points") and should be listed under "Charge Point Overview".

...

#### Actual Behavior
After pushing "Add" in "Unknown Charge Points" for e.g. ChargeBox ID "15999999_19999**.0B**", Charge Point is still listed in this category ("Unknown Charge Points") but also listed unter "Charge Point Overview", but with ChargeBoxID "15999999_19999". The part after "." is missed out.
...

#### Steps to Reproduce the Problem

  1. Set up ChargePoint with ID, containing "." or maybe any other special characters 
  2. Try adding it unter "Unknown Charge Points"

[image]

[image]

goekay, 2020-01-14:
hi, thanks for the detailed issue report. it is a valid bug. i will commit the fix to provide the expected behaviour in a few minutes.

closed 2020-01-14 by goekay

ue-lee, 2020-01-15:
Thanks for the fast patch!

commit 76440311 (2020-01-14): Fixes #270