steve-community/steve #273: data Transfer Implementation

eviontech, 2020-01-16:
#### Checklist

- [ ] I checked other issues already and found no answer
- [ ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve
the vendor specific data Transfer request implementation just dumps the data to the log.
...

#### Describe the solution you'd like
create a new table for transaction data and dump the data received with reference to the transaction id
...

#### Describe alternative solutions or features you've considered 
tried to send the data part from the request to other data dump file from where i can access it for further use
...

#### Additional context

...

eviontech, 2020-01-16:
this can be considered continuation to issue #263

goekay, 2020-01-16:
sorry, but i don't see the value of writing this data into database, because as i wrote in the previous issue, the implementation of this function is just a place holder, it is a dummy implementation. data transfer is an extension of ocpp, in cases a function is needed that is not covered by ocpp. therefore, this is highly charging station vendor-specific. and therefore, the user of steve should adapt/change it as needed.

> create a new table for transaction data and dump the data received with reference to the transaction id

data transfer has nothing to do with transactions or transaction ids.

eviontech, 2020-01-19:
> data transfer has nothing to do with transactions or transaction ids.
we need it based on this so i was just asking if u can guide us with a simple demo function to write the data recieved into a new table

csamsel, 2020-01-20:
not without knowing the vendor specification.

eviontech, 2020-01-20:
[INFO ] 2020-01-06 02:32:45,216 de.rwth.idsg.steve.service.CentralSystemService16_Service - [Data Transfer] Charge point: VNT-HR-001, Vendor Id: VNT-HR
[INFO ] 2020-01-06 02:32:45,216 de.rwth.idsg.steve.service.CentralSystemService16_Service - [Data Transfer] Data: {'transactionId':130,'BatteryCapacity':10000,'RatedVoltage':500,'RatedCurrent':30,'BatteryPresentStatus':90,'AvailableRange':0,'DuringLastCharging':0,'DistanceTravelledSinceLastCharge':0,'LastChargingEndReason':00,'BattetyTemperature':0,'ChargerTemperature':0,'BatteryPackNo':4660,'BatterySetChargingFreq':3,'BatterySerialNumber':20160101,'LastChargeStatus':'0000000000000000','TypeCharger':'SingleCharger','BatteryType':'lithium iron phosphate battery','VINNumber':'0123456789ABCDEFG','Location':'FFF','BatteryManufacturer':'ABCD','ChargerSuspendingChargingInfo':0,'BMSSuspendingChargingInfoa':0}

these are the two info messsages i am trying to save to database

the later message has a json string as data but will need to replace ' with " in java

goekay, 2020-01-20:
@eviontech saving this json data to database is an easy job. sure, i could do it. but saving it alone probably will not be useful for you. you need to process it, extract information from it, and do something with it. here we hit the wall, which i was trying to communicate: this data is specific to your vendor and to your use case and storing it in the database will not be the solution for your problems.

goekay, 2020-01-20:
as for the guidance, you can look into our tools for database: flyway to create and execute migration scripts, jooq for creating type-safe sql queries and executing them. the codebase has many, many uses of jooq in Repository classes. i hope this helps.

closed 2020-01-26 by goekay