steve-community/steve #275: Include local date-time format

fusionJose, 2020-01-24:
Is there a way to send the date-time in local format instead of UTC? There are some servers which send it in this format.

Steve sends: "2020-01-24T09:47:39.340Z"

It would be fine to send "2020-01-24T09:47:39+00:00"

csamsel, 2020-01-24:
if you want to you can change the date formatter in /src/main/java/de/rwth/idsg/steve/utils/DateTimeUtils.java 
but be advised that SteVes behavior is correct.

fusionJose, 2020-01-27:
Can you help me to change the date formatter? I am a bit lost in java. Thank you very much

goekay, 2020-01-27:
there are two arguments in your comment. i don't know which is your actual problem. i will try to address both.

> Is there a way to send the date-time in local format instead of UTC? 

yes. if the problem is only overriding the UTC time zone with your local one, [this](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java#L46) is the place to change the default behaviour. just make the line as follows:

`
private final String timeZoneId = ZoneId.systemDefault().getId();
`

with this, steve will use the time zone of the server it is running in. now, your second argument:

> Steve sends: "2020-01-24T09:47:39.340Z"
> It would be fine to send "2020-01-24T09:47:39+00:00"

i hope you were giving an example for the previous argument. these both expressions are semantically the same and telling the same date/time in the same time zone. i hope that the charging station is not that picky, such that it does not accept a UTC time with `Z` at the end but expects `+00:00` instead. i don't know whether we could and should work around this in steve. if this is the case, it is a bug and you should contact the charging station vendor.

fusionJose, 2020-01-29:
Okey, That's what i needed.

Thank you very much.

closed 2020-01-29 by fusionJose