steve-community/steve #279: Any guidance to add a charge point is appreciated

ijazm123, 2020-01-27:
Need to know how to add a charge pont

csamsel, 2020-01-27:
Feel free to read the documentation: https://github.com/RWTH-i5-IDSG/steve
It's described in detail there.

ijazm123, 2020-01-27:
while adding the charger point I am getting an error 'Registration status is required'

ijazm123, 2020-01-27:
> Feel free to read the documentation: https://github.com/RWTH-i5-IDSG/steve
> It's described in detail there.

while adding the charger point I am getting an error 'Registration status is required'

goekay, 2020-01-27:
> while adding the charger point I am getting an error 'Registration status is required'

this was a bug that was reported and i fixed recently (#248). but i did not make a release after the fix, yet. so, if you are using the latest release version, this error will happen.

i will make a release later based on the latest commit. until then, you can 'git pull' the repo use it.

goekay, 2020-01-27:
i made a new release which includes the fix: https://github.com/RWTH-i5-IDSG/steve/releases/tag/steve-3.4.2

sorry for the inconvenience.

closed 2020-01-27 by goekay

ijazm123, 2020-01-28:
> i made a new release which includes the fix: https://github.com/RWTH-i5-IDSG/steve/releases/tag/steve-3.4.2
> 
> sorry for the inconvenience.

Thank you so much. Now the issue is resolved.