steve-community/steve #280: Question: range of integer types (transactionId, ...)

lategoodbye, 2020-01-29:
Still after reading the OCPP 1.6 specification it's not clear me, what range the several integer types like transactionId, chargeProfileId, reservationId, listVersion have.

Is there some kind of minimum range, which all CP implementations should support?
Should i assume these all can be signed?

csamsel, 2020-01-29:
I agree it's not well defined. Personally i would have assumed 32bit unsigned integer.

lategoodbye, 2020-01-29:
Looking at the OCPP 2.0 spec shows a better definition. The primitive type "integer" is defined as signed 32 bit integer, which is easier in Java. But the seqNo for the transaction should be 32 bit unsigned.

lategoodbye, 2020-01-30:
Except of the specification, it would be nice to know the ranges of the SteVe implementation.

goekay, 2020-01-31:
transactionId, chargeProfileId, reservationId are backed by the database primary key type. therefore, they are auto incrementing, unsigned INT types in MySQL (https://dev.mysql.com/doc/refman/5.7/en/integer-types.html)

listVersion is not persisted in database. the user is responsible to choose it as she sees fit. its data type is an int  in Java (32 bit, signed)

goekay, 2020-02-16:
i consider the question answered and this issue resolved. therefore, closing it.

closed 2020-02-16 by goekay