steve-community/steve #285: Question: Time Response delay

fusionJose, 2020-02-03:
I need to test in a charging station some cases of delay between messages. We actually have some problems due to 3g networks.

So my question is if there is a simple way, in Steve server, to add a delay between the receipt of a websocket message from a charging station and it corresponding response (For a bootnotification request e.g.).

goekay, 2020-02-04:
the easiest way to do this would be to let the thread handling the message sleep just before the response is actually sent. it might have some side effects, though. therefore, handle with care.

[this class](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/pipeline/IncomingPipeline.java) is responsible for sending messages. i created a gist with the necessary modifications. [these](https://gist.github.com/goekay/eb346f228a544d412d00c741d664a594#file-incomingpipeline-java-L65-L70) are the additional lines necessary. change the value 5 to any other desired value.

fusionJose, 2020-02-05:
Thank you very much. 
However there are some unresolved compilation problems with this:

        log cannot be resolved
	The method getProfile() is undefined for the type SteveConfiguration
	The method getTimeZoneId() is undefined for the type SteveConfiguration
	The method getTimeZoneId() is undefined for the type SteveConfiguration
	log cannot be resolved
	The method getTimeZoneId() is undefined for the type SteveConfiguration
	The method getProfile() is undefined for the type SteveConfiguration
	DEV cannot be resolved to a variable
	TEST cannot be resolved to a variable
	PROD cannot be resolved to a variable

Any idea? I don´t understand java very much. Maybe are there some conflicts with the time? when I import java.util.concurrent.TimeUnit

BR

fusionJose, 2020-02-06:
Okey, thank you very.

That's what i needed.

closed 2020-02-06 by fusionJose