steve-community/steve #291: new Box: ABL 3W2260 - communication problems

Jeepersfinest, 2020-02-08:
#### Checklist

- [ x] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.2
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior
In Log of wallbox every minute:
2020-02-08 18:10:04,246 ERROR y.fr [Thread-5] Not supported device type: SMART_METER_GATEWAY

in Steve: Paramter of "meter type" etc are empty.
see pic

[image]

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context
Same with 1.5 and 1.6 OCPP.

Is their any way to get the ABL-Box as a supported device?

billy995, 2020-02-08:
This typ of wallbox has a logging gateway so you can get the meter information from the OCMF dataset normally it is a Carlo Gavazi meter type EM340 with a RS485 modbus connection

Jeepersfinest, 2020-02-08:
Hi, thanks for your support.
You are right, there is a logging gateway. Can i connect it to steve as well?
[image]

billy995, 2020-02-08:
I'm not sure but the OCMF dataset must be saved in mysql database

Jeepersfinest, 2020-02-08:
?

billy995, 2020-02-08:
The wall box communicate in OCPP 1.5 and OCPP 1.6 so it must work but I don't understand why you buy an 3W2260 if you use it at home you just need an box with LGW if you want to use it public or with payment system

csamsel, 2020-02-08:
@Jeepersfinest what the use case? do you want to use the rfid card authorization, remote start charging etc via OCPP?

Jeepersfinest, 2020-02-09:
My use case: I share my box with 2 neighbours. I will use different RFID-Cards. Based on that i need a montly report of the consumption.

On a further perspective i want to:
1. Integrate the box in my home automation
2. when legal questions a solved, integrate the box into the backend of "DKV" - the provide of our fuel billing card...  
 
Hope that helps.

goekay, 2020-02-09:
from the screenshot i can see that the charging station can connect to steve. because otherwise, all of these fields should be empty. so, there are no connectivity or network issues. that's good.

> 2020-02-08 18:10:04,246 ERROR y.fr [Thread-5] Not supported device type: SMART_METER_GATEWAY

i don’t know why this is the case.

> in Steve: Paramter of "meter type" etc are empty.

if the station does not provide these information, they will be empty.

look at the logs of steve to see any funny behaviours or whether there are exceptions.

i don't know how the logging gateway topic is related to your problem. this is an area where i (or we) cannot help you, because it is outside the scope of steve.

your use cases indeed can be realised with steve. we support rfid cards. transactions are stored in database. so you can look at the transactions for a card and for a month period to see how much is consumed.

Jeepersfinest, 2020-02-14:
Hi, 
after a few test i think the LGW can not be accessed by Steve...there is a seprate windows tool to connect by rs485.
But next step is a problem:
Remote Start/stop etc works fine.
i add a user and a tag (same number that is reported in logs), uploaded it to "local list" in my box.
But it doesn´t start charging...
Log of my box:
2020-02-14 13:19:02,290 INFO  y.T [Event Dispatch] authentication response, EVSE #1, accepted=false
--
2020-02-14 13:19:02,144 INFO  y.bV  [Event Dispatch] IdToken is unknown, expired or not accecpted in  cfg/userLists_backend/localList
2020-02-14 13:19:02,133 INFO  y.bV  [Event Dispatch] IdToken is unknown, expired or not accecpted in  cfg/userLists_backend/localCache

Did i do something wrong?

Jeepersfinest, 2020-02-15:
Problem found ...tags start with 0000...i have to delete them.
Works now fine!

closed 2020-02-15 by Jeepersfinest