steve-community/steve #292: Transactions never stop

komatar, 2020-02-10:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     :  3.4.1
Operating system  : Windows 10 Pro
JDK               :  11.0.5
Database          : MySQL 5.7.28
```

#### Expected Behavior

Transaction stops after car is unplugged.

#### Actual Behavior

Transaction is active after the car is unplugged, until it's manually stopped. 

#### Steps to Reproduce the Problem

  1. Plug in car
  2. Wait
  3. Plug out car

#### Additional context

I'm using charging station listed in the Wiki as compatible: Keba KeContact P30, though a newer version of firmware and SteVe.

When I manually stop it, the start_value and stop_value are correct, while the stop_timestamp is rounded to 15 minutes, which I assume is from the charging station polling settings.
When I manually stop it, it gets the next incremental `transaction_stop.transaction_pk` number. In `transaction_stop_failed.transaction_pk`, all of them are -1.

Below is a `fail_reason` from table `transaction_stop_failed`.
```
org.jooq.exception.DataAccessException: SQL [insert into `ocppserver`.`transaction_stop` (`transaction_pk`, `event_timestamp`, `event_actor`, `stop_timestamp`, `stop_value`, `stop_reason`) values (?, ?, ?, ?, ?, ?)]; Data truncation: Out of range value for column 'transaction_pk' at row 1
	at org.jooq_3.12.3.MYSQL.debug(Unknown Source)
	at org.jooq.impl.Tools.translate(Tools.java:2717)
	at org.jooq.impl.DefaultExecuteContext.sqlException(DefaultExecuteContext.java:755)
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:382)
	at org.jooq.impl.AbstractDelegatingQuery.execute(AbstractDelegatingQuery.java:119)
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.updateTransaction(OcppServerRepositoryImpl.java:271)
	at de.rwth.idsg.steve.service.CentralSystemService16_Service.stopTransaction(CentralSystemService16_Service.java:216)
	at de.rwth.idsg.steve.ocpp.soap.CentralSystemService15_SoapServer.lambda$stopTransaction$6(CentralSystemService15_SoapServer.java:132)
	at java.base/java.util.function.Function.lambda$andThen$1(Function.java:88)
	at java.base/java.util.function.Function.lambda$andThen$1(Function.java:88)
	at de.rwth.idsg.steve.ocpp.soap.CentralSystemService15_SoapServer.stopTransaction(CentralSystemService15_SoapServer.java:134)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.apache.cxf.service.invoker.AbstractInvoker.performInvocation(AbstractInvoker.java:179)
	at org.apache.cxf.jaxws.JAXWSMethodInvoker.performInvocation(JAXWSMethodInvoker.java:66)
	at org.apache.cxf.service.invoker.AbstractInvoker.invoke(AbstractInvoker.java:96)
	at org.apache.cxf.jaxws.AbstractJAXWSMethodInvoker.invoke(AbstractJAXWSMethodInvoker.java:232)
	at org.apache.cxf.jaxws.JAXWSMethodInvoker.invoke(JAXWSMethodInvoker.java:85)
	at org.apache.cxf.service.invoker.AbstractInvoker.invoke(AbstractInvoker.java:74)
	at org.apache.cxf.interceptor.ServiceInvokerInterceptor$1.run(ServiceInvokerInterceptor.java:59)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at org.apache.cxf.interceptor.ServiceInvokerInterceptor$2.run(ServiceInvokerInterceptor.java:126)
	at org.apache.cxf.workqueue.SynchronousExecutor.execute(SynchronousExecutor.java:37)
	at org.apache.cxf.interceptor.ServiceInvokerInterceptor.handleMessage(ServiceInvokerInterceptor.java:131)
	at org.apache.cxf.phase.PhaseInterceptorChain.doIntercept(PhaseInterceptorChain.java:308)
	at org.apache.cxf.transport.ChainInitiationObserver.onMessage(ChainInitiationObserver.java:121)
	at de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor.handleMessage(MediatorInInterceptor.java:95)
	at de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor.handleMessage(MediatorInInterceptor.java:50)
[... 32 lines cut ...]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103)
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126)
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:781)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:917)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: com.mysql.cj.jdbc.exceptions.MysqlDataTruncation: Data truncation: Out of range value for column 'transaction_pk' at row 1
	at com.mysql.cj.jdbc.exceptions.SQLExceptionsMapping.translateException(SQLExceptionsMapping.java:104)
	at com.mysql.cj.jdbc.ServerPreparedStatement.serverExecute(ServerPreparedStatement.java:637)
	at com.mysql.cj.jdbc.ServerPreparedStatement.executeInternal(ServerPreparedStatement.java:418)
	at com.mysql.cj.jdbc.ClientPreparedStatement.execute(ClientPreparedStatement.java:370)
	at com.zaxxer.hikari.pool.ProxyPreparedStatement.execute(ProxyPreparedStatement.java:44)
	at com.zaxxer.hikari.pool.HikariProxyPreparedStatement.execute(HikariProxyPreparedStatement.java)
	at org.jooq.tools.jdbc.DefaultPreparedStatement.execute(DefaultPreparedStatement.java:209)
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:450)
	at org.jooq.impl.AbstractDMLQuery.execute(AbstractDMLQuery.java:874)
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:368)
	... 70 more

```

goekay, 2020-02-13:
> In transaction_stop_failed.transaction_pk, all of them are -1.

i think this value here is the problem. if you start the transaction normally, the station receives the transaction id generated by steve (transaction_pk in table is the transaction id). after that, when stopping, the station should use the transaction id assigned to this charging session. apparently, it sends the stop transaction message with -1, which is not normal and hence the exception.

goekay, 2020-03-25:
closing due to inactivity.

closed 2020-03-25 by goekay