steve-community/steve #296: Jason- Connection could not established

Jeepersfinest, 2020-02-15:
#### Checklist

- [ x] I checked other issues already, but found no answer/solution
- [ x] I checked the documentation and wiki, but found no answer/solution
- [ x] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.2
Operating system  : ...
JDK               : ...
Database          : ...

OCPP 1.6s

#### Expected Behavior
Switching to Jason -> no connection to steve
SOAP works fine

#### Actual Behavior
When i use this link to switch to WS: 
ws://192.168.100.9:8080/steve/websocket/CentralSystemService/ABL3W22600xxxx
the connection could not established:
2020-02-15 14:54:08,156 ERROR y.du [WebsocketHandler] TyrusClientEngineState(org.glassfish.tyrus.core.HandshakeException): Response header is wrong.

what do i wrong?

And 1 additinal question:
My box supports WS with and without TLS.
Does STEVE supports WSS?

Thanks!
Christian

goekay, 2020-02-16:
steve supports `ws` and `wss` both.

> 2020-02-15 14:54:08,156 ERROR y.du [WebsocketHandler] TyrusClientEngineState(org.glassfish.tyrus.core.HandshakeException): Response header is wrong.

this here might be a hint. but i don't know why it happens. a more detailed report would be necessary.

Jeepersfinest, 2020-02-16:
Hi,
i get a step further. ws works now.
I have to use: ws://192.168.100.9:8080/steve/websocket/CentralSystemService/
(without Charge-ID) - my box adds it.

a last question :-)
I want to use wss, but how can i activate it on the Steve-Server?
Do i need a certificate? Where can i find it?

Thanks a lot!

goekay, 2020-02-16:
> Do i need a certificate?

Not necessarily. Java comes with some built-in certificates. If your station uses a certificate, that can be trusted by the certificate chain of Java, then you can just use it without additional configuration.

If the station uses a custom, self-signed certificate, that cannot be trusted in the certificate chain, you should add this to the java keystore of Steve and enable HTTPS (please look at our wiki).

Probably it will work without custom config. Just try it.

Jeepersfinest, 2020-02-18:
Hi, i´m a noob in SSL and certificates etc.
I followed the path in wiki, create a certificate on Steve....and now?
Do i have to importe that in my station or do i have to import the certification of my station into the java store?

If a change ws into wss, thet state (in my station) changes into "backend does not accept connection". Steve-Gui is accessible bei HTTPS - works fine.

Thanks for a hint!

goekay, 2020-02-20:
> If a change ws into wss, thet state (in my station) changes into "backend does not accept connection".

Please post the part of the log file of Steve for this particular event.

Jeepersfinest, 2020-02-20:
Thanksagain for your support!
Station: 

2020-02-20 19:42:04,946 INFO  y.au [WebsocketHandler] overall connectivity state=SECURE_MISSING_BACKEND_CERTIFICATE
--
2020-02-20 19:42:04,944 INFO  y.cS [OCPP Request Executor] received boot notification response: null
2020-02-20 19:42:04,900 ERROR y.du  [WebsocketHandler]  SunCertPathBuilderException(sun.security.provider.certpath.SunCertPathBuilderException):  The certificate of Backend was not found. Trying disconti
2020-02-20 19:42:04,896 ERROR y.du  [WebsocketHandler] SSLHandshake(javax.net.ssl.SSLHandshakeException):  General SSLEngine problem. Trying will be discontinued.

In Steve-log is nothing to see.

Jeepersfinest, 2020-02-29:
any ideas?

Jeepersfinest, 2020-02-29:
I found a solution!
Export of certificate on Steve and import that in my station, change to wss and reboot. Ready!

closed 2020-02-29 by Jeepersfinest