steve-community/steve #299: Reservation with start and stop date-time in testing

naveen2189, 2020-02-24:
#### Checklist

- [x] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [3.4.1] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

Need reservation testing for a limited period by using start and stop with respective date and time. Since if the charger is live unable to reserve for later days.

#### Describe the solution you'd like

Reservation with start and stop date-time in testing

#### Describe alternative solutions or features you've considered 

Nil

#### Additional context

...

goekay, 2020-02-27:
the reservation operation in ocpp defines a reservation period starting from current date/time to an expiry date/time. therefore, the reservations always start with _now_.

naveen2189, 2020-02-27:
Then how to configure in mobile app before reaching the charging station?---- On Thu, 27 Feb 2020 19:02:31 +0000  notifications@github.com<notifications@github.com> wrote ----the reservation operation in ocpp defines a reservation period starting from current date/time to an expiry date/time. therefore, the reservations always start with now.

goekay, 2020-02-27:
reservations far in the future (be it tomorrow, next week, month or whatever your horizon is) are not part of the ocpp and therefore steve. if you have a such a use case, you can implement it on top of steve.

goekay, 2020-03-25:
closing due to inactivity.

closed 2020-03-25 by goekay