steve-community/steve #306: Unable to run server

mkj-git, 2020-03-08:
I am trying run steve server. When i am open http://127.0.0.1:8080/steve/manager/signin i am getting 

```
HTTP ERROR 404
Problem accessing /steve/WEB-INF/views/signin.jsp. Reason:

    /steve/WEB-INF/views/signin.jsp
```
In eclipse console it shows:

```
Log file: C:\Users\manishk\logs\steve.log
Starting.......... Done!
Hint: You can stop the application by pressing CTRL+C

Access the web interface using
- http://127.0.0.1:8080/steve/manager/home
SOAP endpoint for OCPP
- http://127.0.0.1:8080/steve/services/CentralSystemService
WebSocket/JSON endpoint for OCPP
- ws://127.0.0.1:8080/steve/websocket/CentralSystemService/(chargeBoxId)

```

Here is log file content:

```
[INFO ] 2020-03-08 20:03:01,167 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2020-03-08 20:03:01,621 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2020-03-08 20:03:01,636 de.rwth.idsg.steve.Application - Date/time zone of the application is set to UTC. Current date/time: 2020-03-08T14:33:01.621Z
[INFO ] 2020-03-08 20:03:01,657 org.eclipse.jetty.util.log - Logging initialized @1306ms to org.eclipse.jetty.util.log.Slf4jLog
[INFO ] 2020-03-08 20:03:01,837 org.eclipse.jetty.server.Server - jetty-9.4.19.v20190610; built: 2019-06-10T16:30:51.723Z; git: afcf563148970e98786327af5e07c261fda175d3; jvm 11.0.4+10-LTS
[INFO ] 2020-03-08 20:03:01,937 org.eclipse.jetty.server.session - DefaultSessionIdManager workerName=node0
[INFO ] 2020-03-08 20:03:01,937 org.eclipse.jetty.server.session - No SessionScavenger set, using defaults
[INFO ] 2020-03-08 20:03:01,937 org.eclipse.jetty.server.session - node0 Scavenging every 600000ms
[INFO ] 2020-03-08 20:03:01,937 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring root WebApplicationContext
[INFO ] 2020-03-08 20:03:01,937 org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization started
[INFO ] 2020-03-08 20:03:02,639 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
[INFO ] 2020-03-08 20:03:02,786 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
[INFO ] 2020-03-08 20:03:05,345 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2020-03-08 20:03:05,731 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP12
[INFO ] 2020-03-08 20:03:05,746 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2012/06/}CentralSystemService from class ocpp.cs._2012._06.CentralSystemService
[INFO ] 2020-03-08 20:03:05,793 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP15
[INFO ] 2020-03-08 20:03:05,793 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2015/10/}CentralSystemService from class ocpp.cs._2015._10.CentralSystemService
[INFO ] 2020-03-08 20:03:05,828 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP16
[INFO ] 2020-03-08 20:03:05,830 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2020-03-08 20:03:05,835 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemService
[INFO ] 2020-03-08 20:03:07,114 org.hibernate.validator.internal.util.Version - HV000001: Hibernate Validator 6.1.2.Final
[INFO ] 2020-03-08 20:03:07,353 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: Ant [pattern='/static/**'], []
[INFO ] 2020-03-08 20:03:07,353 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: Ant [pattern='/views/**'], []
[INFO ] 2020-03-08 20:03:07,368 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: any request, [org.springframework.security.web.context.request.async.WebAsyncManagerIntegrationFilter@1fe37d27, org.springframework.security.web.context.SecurityContextPersistenceFilter@610d0067, org.springframework.security.web.header.HeaderWriterFilter@7f9b851c, org.springframework.security.web.csrf.CsrfFilter@62376bdd, org.springframework.security.web.authentication.logout.LogoutFilter@64245ca9, org.springframework.security.web.authentication.UsernamePasswordAuthenticationFilter@7ec75228, org.springframework.security.web.authentication.www.BasicAuthenticationFilter@3a720ae3, org.springframework.security.web.savedrequest.RequestCacheAwareFilter@7a1878d, org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter@35a2ea5a, org.springframework.security.web.authentication.AnonymousAuthenticationFilter@1718de70, org.springframework.security.web.session.SessionManagementFilter@4d529bc0, org.springframework.security.web.access.ExceptionTranslationFilter@4afdb05e, org.springframework.security.web.access.intercept.FilterSecurityInterceptor@6164b3a2]
[INFO ] 2020-03-08 20:03:07,538 org.springframework.web.context.ContextLoader - Root WebApplicationContext initialized in 5601 ms
[INFO ] 2020-03-08 20:03:07,538 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring DispatcherServlet 'spring-dispatcher'
[INFO ] 2020-03-08 20:03:07,538 org.springframework.web.servlet.DispatcherServlet - Initializing Servlet 'spring-dispatcher'
[INFO ] 2020-03-08 20:03:07,553 org.springframework.web.servlet.DispatcherServlet - Completed initialization in 15 ms
[INFO ] 2020-03-08 20:03:07,569 org.eclipse.jetty.server.handler.ContextHandler - Started o.e.j.w.WebAppContext@5cef5fc9{/steve,file:///D:/dev/workspaces/steve-master/steve-master/target/classes/webapp/,AVAILABLE}
[INFO ] 2020-03-08 20:03:07,584 org.eclipse.jetty.server.AbstractConnector - Started ServerConnector@738153d8{HTTP/1.1,[http/1.1]}{127.0.0.1:8080}
[INFO ] 2020-03-08 20:03:07,584 org.eclipse.jetty.server.Server - Started @7234ms

```

mkj-git, 2020-03-08:
I had commented following line:

```
                <executions>
                    <execution>
                        <phase>generate-sources</phase>
                        <goals>
                            <goal>migrate</goal>
                        </goals>
                    </execution>
                </executions>
```

removing the comment solved it.

goekay, 2020-03-08:
> In eclipse console it shows:

i have a feeling that you are building and starting the project from within eclipse. please do not do it. i would advise to use the maven commands on command line. 

there is nothing wrong the lines that you commented out. i stopped using eclipse because of such buggy maven integration problems.

mkj-git, 2020-03-09:
Yes I agree. In eclipse, i can see some RED line indication by eclipse because its failing to resolve some dependency. But when i run it its running without any issues.

goekay, 2020-03-21:
i consider this issue resolved and am closing it.

closed 2020-03-21 by goekay