steve-community/steve #310: OCPP 1.6 - Watts vs Amperes in a charging profile

fassedou, 2020-03-20:
Hello,

This issue concerns OCPP 1.6.

When creating a charging profile on SteVe and adding a schedule period, the limit seems to be required to be set in amperes (the field name is "power limit (in amperes)". This doesn't seem to follow the OCPP 1.6 standard because the limit can be set in Watts, depending on the value of "ChargingRateUnit".

I assume the problem was that in the first version of the 1.6 standard (in 2015), OCA said that indeed the limit had to be set in amperes and that "ChargingRateUnit" is only used for "MinChargingRate". However, this was a mistake from the OCA and they published in 2018 a second version of the 1.6 standard with an erratum that stated that the limit can actually be set in watts depending on the value of "ChargingRateUnit".

Thanks

goekay, 2020-03-25:
hello @fassedou, this is very valuable input. thanks! i think your assumption is correct.

on the bright side, i think only the description on the page "power limit (in amperes)" is misleading. the implementation does not differentiate between ampere or watt, but handles the value just as a value and sets it.

but i will do a more thorough investigation and update on this ticket.

goekay, 2021-04-14:
the investigation of @fassedou is correct. the spec version we used with release date `2015-10-08` had the following description for this field:
[image]

with `edition 2 FINAL, 2017-09-28`, this was corrected in the errata sheet as described:
[image]

as far as steve is concerned, this implies 1) renaming db column 2) renaming java fields/methods and 3) cosmetic changes on web pages. functionality is not impacted. i am working on a fix.

closed 2021-04-14 by goekay