steve-community/steve #318: Illegal reflective access operation on JDK 11 with SteVe 3.4.2

dhanushkatennakoon, 2020-04-01:
SteVe Version         : 3.4.2
Operating system  : Windows 10
JDK                        : 11.0.6
Database               : MYSQL 5.7

Hi Team, 

I have successfully built SteVe. But when I run SteVe it comes with an error saying "Illegal reflective access by org.jooq.tools.reflect.Reflect" and from there it fails running SteVe.

I can build SteVe successfully with "mvn package" and then when I SteVe using "java -jar target/steve.jar" below error message comes and prevents SteVe from running.

Error message; 
"Starting..............WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.jooq.tools.reflect.Reflect (file:/C:/Users/chanu/steve-steve-3.4.2/target/libs/jooq-3.12.3.jar) to constructor java.lang.invoke.MethodHandles$Lookup(java.lang.Class)
WARNING: Please consider reporting this to the maintainers of org.jooq.tools.reflect.Reflect
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
................. FAILED!
Please refer to the log file for details"

Googling it it appears that this is the common issue with JOOQ 3.12.3 and JDK 11.  and was fixed in JOOQ 3.13.0.

My question is have others faced the same issue with SteVe 3.4.2 which uses JOOQ 3.12.3? Is there a fix for that please?

many thanks all! Much appreciated on your help on this.
Dan

goekay, 2020-04-01:
hello, we had similar questions before. the warnings are not a problem and the reason for the failure. they are just warnings (search for closed issues). the reason why steve startup failed can be found in the log file as indicated in the terminal output.

dhanushkatennakoon, 2020-04-03:
> hello, we had similar questions before. the warnings are not a problem and the reason for the failure. they are just warnings (search for closed issues). the reason why steve startup failed can be found in the log file as indicated in the terminal output.

Hi goekay, Many thanks for your quick response. I have been working on that issue on why i can't run SteVe after successfully building it in mvn. But still can't fix it. Hope you can help.

Log file shows only one error, which is "MySQL and Java are not using the same time zone. Java offset in seconds (0) != MySQL offset in seconds (3600)". I'm UK based and mySQL uses system time which is UTC and SteVe default for Java is also UTC, So why do i have that error and how can i fix it please?

Strangely enough i manage to get SteVe running in Feb 2020 and that stopped running hence i am trying to get SteVe build and run again. Why my previously successfully ran SteVe and new Steve is not running? Is that something to do with the Summer Time adjustment?

Sorry if this is such a basic question. Much appreciated your help on this.

dhanushkatennakoon, 2020-04-04:
This now closed.Issue was caused by day light saving 1 hr difference which caused mySQL and Java timing to be different. Resolved by changing MYSQL timezone by
SET @@global.time_zone = '+00:00';

and issue is resolved and SteVe runs perfectly. Many thanks for SteVe and goekay for giving his insight quickly. Thanks all!

closed 2020-04-04 by dhanushkatennakoon

goekay, 2020-04-04:
> SET @@global.time_zone = '+00:00';

please keep in mind, this only affects the running mysql instance. if you stop and restart mysql, it will use the time zone setting from configuration. the best approach is to set it in the configuration file of mysql.