steve-community/steve #327: Connector Status for 1 & 2 are not being updated

GRangaRao, 2020-04-23:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...3.4.1
Operating system  : ...18.04 ubuntu
JDK               : ...Open JDK 11
Database          : ...MySQL 5.7
```

#### Expected Behavior
Connector Status Expected to be activated for each connector

Left and Right connectors or  1 & 2
...

#### Actual Behavior

Steve Dashboard only shows connector 0, Not showing connector 1 & 2
...

#### Steps to Reproduce the Problem

  1. ...Boot the Charge Station
  2. ...Settings-> Change Availability-> Select Charge Station, Connector ID, Select 
[image]
Operative
  3. ...

#### Additional context

...

csamsel, 2020-04-23:
operational status is sometimes only provided for the whole charging station, not per connector. you can check the log file to see what the charging station actually sent to the backend.

goekay, 2020-04-23:
connector id 0 represents the whole station and steve, therefore, gives it priority if its timestamp is newer than the particular connectors.

GRangaRao, 2020-04-23:
Thanks for your comments csamsel & goekay, am looking for both connectors to be seen on steve. 
Charge Station need to send the connector ID back to OCPP server steve to show both connectors?

goekay, 2020-04-25:
yes. it probably already sent the connector ids. you can look at the logs. the distinction we are making is: if the message timestamp for 0 is more recent than the message timestamp of other connectors, steve only shows the status of 0. if you have database knowhow, you can look at the "connector" table for the connectors of charging stations.

goekay, 2020-05-02:
since there was no further activity, i consider this issue resolved.

closed 2020-05-02 by goekay