steve-community/steve #333: Ocpp 1.6 Error codes, Steve has implementation to push these Error info?

GRangaRao, 2020-04-28:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

My query is not related to a bug, but to ensure all Errors listed in Ocpp 1.6 spec are logged into connector_status table?

Connector_status table has 4 fields
error_code | error_info | vendor_id | vendor_error_code

Please confirm whether all the Errors would be logged into the connector_status table fields?

Error definitions given in the spec
1.ConnectorLockFailureFailure to lock or unlock connector.
2.EVCommunicationErrorCommunication failure with the vehicle, might beMode 3 or other communication protocolproblem. 
  This is not a real error in the sense thatthe Charge Point doesn’t need to go to the 3.faultedstate. Instead, it should go to the SuspendedEVSEstate.
4.GroundFailureGround fault circuit interrupter has beenactivated.
5.HighTemperatureTemperature inside Charge Point is too high.
6.InternalErrorError in internal hard- or software component.
7.LocalListConflictThe authorization information received from theCentral System is in conflict with theLocalAuthorizationList.
8.NoErrorNo error to report.
9.OtherErrorOther type of error. More information invendorErrorCode.
10.OverCurrentFailureOver current protection device has tripped.
11.OverVoltageVoltage has risen above an acceptable level.
12.PowerMeterFailureFailure to read power meter.
13.PowerSwitchFailureFailure to control power switch.
14.ReaderFailureFailure with idTag reader.
15.ResetFailureUnable to perform a reset.
16.UnderVoltageVoltage has dropped below an acceptable level.
17.WeakSignalWireless communication device reports a weaksignal
...

Thank you for your time, Appreciate your answers and responses

goekay, 2020-04-28:
> Please confirm whether all the Errors would be logged into the connector_status table fields?

yes. the error_code column stores this info.

goekay, 2020-05-02:
i consider the question answered and this issue resolved.

closed 2020-05-02 by goekay