steve-community/steve #342: My SQL 8.x & steve?

GRangaRao, 2020-05-08:
Is there a reason why Steve is not using My sql 8?

ReadMe

* At least MySQL 5.7.7 (MariaDB 10.2.1 or later works as well) as database (**Note: MySQL 8.x is not supported yet**)

Anyone tested MySQL 8.0 with steve ocpp_tables?

csamsel, 2020-05-08:
Because it doesnt work. 

@goekay i think we have to have a look at this, as mysql 8 became the default for Ubuntu 20.04 LTS.

goekay, 2020-05-08:
AFAIK, there is a closed issue where someone tried to use it, but failed due to some changes in 8.0 and our migrations could not be applied. i could not find the issue now.

csamsel, 2020-05-08:
Let me do some test as well and extend the Travis cases and see how it goes.

goekay, 2020-05-09:
i am testing locally with mysql 8.0.19 on mac os x. i did not succeed so far, but here is my report.

1. [database preparation in our readme](https://github.com/RWTH-i5-IDSG/steve#configuration-and-installation) has to be modified slightly:

```
CREATE DATABASE stevedb CHARACTER SET utf8 COLLATE utf8_unicode_ci;
CREATE USER 'steve'@'localhost' IDENTIFIED BY 'changeme';
GRANT ALL PRIVILEGES ON stevedb.* TO 'steve'@'localhost';
-- GRANT SELECT ON mysql.proc TO 'steve'@'localhost';
```

2. the query with `grant select` is commented out because `mysql.proc` does not exist anymore. i don't know the consequences of this one yet, because i could not build steve successfully (see my next point). but as i remember correctly, jooq needed this to read all table info from db.

3. `mvn clean compile` fails, because migrations could not be applied:
```
[ERROR] SQL State  : HY000
[ERROR] Error Code : 1419
[ERROR] Message    : You do not have the SUPER privilege and binary logging is enabled (you *might* want to use the less safe log_bin_trust_function_creators variable)
[ERROR] Location   : /Users/sevketgokay/Documents/git/steve/src/main/resources/db/migration/V0_6_6__inital.sql 
```

goekay, 2020-05-09:
4. if i add the following step in db preparation, migrations work, steve compiles and runs:

`GRANT SUPER ON *.* TO 'steve'@'localhost';`

but this is a global privilege and not a db-level one. therefore, it might come with security concerns.

GRangaRao, 2020-05-11:
Thanks goekay, Lovely to hear, will test and check

goekay, 2020-06-06:
with 3c7319767974f2cfa2ba3d74933a4b35f03e4122 i updated the readme to include the statements from above. therefore, i consider this issue resolved and am closing it.

closed 2020-06-06 by goekay