steve-community/steve #345: Current configuration value under "Change Configuration"

lategoodbye, 2020-05-09:
#### Checklist

- [X] I checked other issues already and found no answer
- [X] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

While editing comma separated lists in "Change Configuration" a user usually needs to know the current value first. So the user is forced to open a second tab to fetch the current values via "Get Configuration". This is cumbersome and error-prone.

#### Describe the solution you'd like

It would be a cool feature to have some kind of button (or automatically on selection) to fetch the current value of selected configuration key into a separate field under the "Change configuration" page.

goekay, 2020-05-10:
current "change configuration" implementation allows to change config of multiple stations with one UI operation. with your proposal, this would be possible no more, though.

lategoodbye, 2020-05-10:
I must confess that i didn't noticed the possibility to change multiple charge points. So for sure this feature request doesn't fit in this case. Unfortunately i don't have an alternative solution.

closed 2020-05-10 by lategoodbye