steve-community/steve #346: Configuration templates

lategoodbye, 2020-05-09:
#### Checklist

- [X] I checked other issues already and found no answer
- [X] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

While provisioning multiple charge points or preparing different test cases the changing of many configuration values is error-prone. It would be nice to prepare a set of config parameter with fixed values and submit them with a single click.

#### Describe the solution you'd like

I've something similar to the "Charging Profile" form in my mind. So the user can create templates with a short text description and assign multiple config parameter there. In order to keep it simple the order shouldn't matter and in case a ChangeConfiguration has been rejected the rest should be dropped.

goekay, 2020-05-10:
hmm... i see the value in this. however, i am hesitant to store these templates in db (as we do for charging profile). i want to change the db schema only if really necessary. maybe we can do this in the java application: as long as the app runs, these are kept in memory. if you stop/restart steve, they will be gone. 

as far as how UI goes.. on the ChangeConfiguration page, we could have another button (beside Perform) to save the configuration that the person just entered.

lategoodbye, 2020-05-10:
This sounds good.