steve-community/steve #350: adding a OCPP Tag List does not work

bkisters, 2020-05-14:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.3
Operating system  : ubuntu 20.04
JDK               : openjdk 11.0.7
Database          :  mariadb  10.3
```

#### Expected Behavior
Import full list of transponder ids
#### Actual Behavior
Error:

org.jooq.exception.DataAccessException: SQL [insert into `stevedb`.`ocpp_tag` (`id_tag`) values (?)]; Duplicate entry '995C65BB' for key 'idTag_UNIQUE'
Reason:

java.sql.BatchUpdateException: Duplicate entry '995C65BB' for key 'idTag_UNIQUE'

#### Steps to Reproduce the Problem

  1. go to ocpp tag list 
  2. insert list of  tags seperated  by newline
  3. click add all

#### Additional context

...

goekay, 2020-05-14:
as the exception states, you already have an idTag '995C65BB' in the table and trying to insert another one.

bkisters, 2020-05-14:
yes, it seems as if there is already that idTag within steve, but I can't find it in the list of tags (datamanagement > ocpp tags )

goekay, 2020-05-14:
maybe you have it twice in the list that you are trying to insert?

bkisters, 2020-05-14:
no. this is the list i actually trying to insert

A1F4FD89
016B65BB
995C65BB
is there a sql table I can search it for?

goekay, 2020-05-14:
of course: `stevedb.ocpp_tag`

bkisters, 2020-05-14:
ok, it is in that table
and now?

goekay, 2020-05-14:
well.. do not insert it twice.

closed 2020-05-14 by bkisters