steve-community/steve #353: SteVe does not read old transactions

bkisters, 2020-05-18:
I was able to successfully read out all transactions of a Mennekes Gatway via a test installation of SteVe.
Now I have completely rebuilt SteVe again (also new database). Unfortunately, SteVe now only reads the new transactions.
SteVe does not read old transactions anymore. Why is that?

goekay, 2020-05-19:
> Now I have completely rebuilt SteVe again (also new database)

since you recreated the database, the old database with all of its information is gone, deleted. therefore, previous transactions are deleted. that's why you cannot see/find them.

bkisters, 2020-05-19:
I thought transactions would be retrieved again from the ChargePoint (Mennekes Gateway) during a new installation from SteVe

goekay, 2020-05-19:
no, it does not happen.

closed 2020-05-19 by bkisters

goekay, 2020-05-19:
in the recent days you opened many issues that were unrelated to an actual problem in steve. please keep in mind, that we do not provide any customer support. github issues are not a support channel.