steve-community/steve #354: SMTP email settings

GRangaRao, 2020-05-23:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

GRangaRao, 2020-05-25:
@csamsel 
@goekay 

Could you please answer this query on smtp email?

csamsel, 2020-05-25:
The error message is clear, the mail server is not reachable.

GRangaRao, 2020-05-25:
Hi csamsel, Mail could be sent with another smtp server by using the same credentials, steve smtp is not working for any reason? is there any configuration option that is missed to add?

goekay, 2020-05-25:
hi @GRangaRao, the fact that another server (https://www.gmass.co/smtp-test) can reach your smtp server does not mean that the server, where steve is deployed, can reach your smtp server. i would suggest to check the network settings.

csamsel, 2020-05-25:
As @goekay said.
Could be an outgoing packet filter preventing that the server can send an email.

GRangaRao, 2020-05-27:
@goekay , @csamsel

server where steve is deployed, i could send the e-mails through smtp with gmass.co/smt-test.

ufw status in ubuntu has port 587 open 

8080/tcp                    ALLOW       Anywhere
25/tcp (v6)                 ALLOW       Anywhere (v6)
587 (v6)                     ALLOW       Anywhere (v6)
587/tcp (v6)               ALLOW       Anywhere (v6)

Still some settings needs correction, as steve could not send emails through SMTP
seeing connection timeout

Thanks for your time!

csamsel, 2020-05-27:
I have nothing to add. If you would try to send an mail from the server where steve is running you would see it isnt working.

btw. the ufw output has the ports only listed as open in IPv6.

closed 2020-05-27 by csamsel