steve-community/steve #355: Mark not connected charge points as unav

redhell, 2020-05-23:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

...

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

redhell, 2020-05-23:
[image]

csamsel, 2020-05-23:
Sending a heartbeat is only a recommendation so we cannot rely on an absence to determine that a charging station is offline.

redhell, 2020-05-24:
The stations are connected via OCPP 1.6J. So it makes sense if WS connection is lost and/or last heartbeat is <time T> ago, the station should have state Unavailable. 
According to the OCPP specs it is advised to send a Heartbeat at least every 24 hours or configurable time interval.

goekay, 2020-05-25:
i am not sure about this. i see the benefit, but it would be a major re-interpretation of ocpp status values. on this table, we are only reading and showing the status values that are sent with the ocpp message. no intermediate magic, no re-interpretation, no "marking a json station with the state unavailable, if it is disconnected"

redhell, 2020-05-29:
@goekay So I should implement this in the controller/service which handles the web socket connections? Can you give me a hint where to go?

csamsel, 2020-05-29:
I would argue that a disrupted web socket connection alone is not indicative that the station is offline. We should only react on a missed heartbeat. So we would need to compare current time > previous heartbeat + interval -> change state in database to something like heartbeat missed.

lategoodbye, 2020-05-30:
Since charge points can operate while been offline so the term "Unavailable" won't be correct from CP perspective. I think it would be better use "Unknown", but this isn't an official status.

I agree with @csamsel that web socket connection shouldn't be the indication for offline in order to handle the SOAP connections, too. But the implementation must compare against the last received PDU, not just Heartbeat.

Beside that, may be this should be (also) a email notification.

goekay, 2020-05-30:
maybe marking json stations as "not connected" additional to the ocpp status is the way to go? i am thinking an asterisk or superscript or something like that.

csamsel, 2020-05-30:
Yeah. Make the state an additional column/field independent of the OCPP state sounds reasonable.

goekay, 2022-02-12:
hey all, i have a working implementation of this feature in the referenced branch. i added an offline icon next to the charging station with a tool tip as shown in the screenshot below.

if you are okay with this, i will merge it later.

[image]

closed 2022-02-13 by goekay

huzaifamk, 2023-07-25:
@goekay if you can share a little that how are you detecting that `conncectors might be not up-to-date`, it will help in implementing some functionality because I have been working with Steve and OCPPs and here is only one hurdle that I am facing that is to detect that `conncectors might be not up-to-date`, is there a way I can see anything related to this in Steve, or there is a formula above, are you using this:  current time > previous heartbeat + interval -> change state in database to something like heartbeat missed?