steve-community/steve #356: Send local list - DIFFERENTIAL

ShadElbo, 2020-06-01:
select one charging station
set list version to "0" or "1" or any number
select "DIFFERENTIAL" update type
select one item from Add/Update list
submit
following error appears:
[ERROR] 2020-06-01 12:45:12,800 de.rwth.idsg.steve.web.GlobalControllerAdvice - Request: http://192.168.0.13:9000/steve/manager/operations/v1.6/SendLocalList raised following exception.
org.jooq.exception.DataAccessException: SQL [select `stevedb`.`ocpp_tag_activity`.`ocpp_tag_pk`, `stevedb`.`ocpp_tag_activity`.`id_tag`, `stevedb`.`ocpp_tag_activity`.`parent_id_tag`, `stevedb`.`ocpp_tag_activity`.`expiry_date`, `stevedb`.`ocpp_tag_activity`.`max_active_transaction_count`, `stevedb`.`ocpp_tag_activity`.`note`, `stevedb`.`ocpp_tag_activity`.`active_transaction_count`, `stevedb`.`ocpp_tag_activity`.`in_transaction`, `stevedb`.`ocpp_tag_activity`.`blocked` from `stevedb`.`ocpp_tag_activity` where `stevedb`.`ocpp_tag`.`id_tag` in (?)]; Unknown column 'stevedb.ocpp_tag.id_tag' in 'where clause'
	at org.jooq_3.12.1.MYSQL.debug(Unknown Source) ~[?:?]
	at org.jooq.impl.Tools.translate(Tools.java:2717) ~[jooq-3.12.1.jar:?]
	at org.jooq.impl.DefaultExecuteContext.sqlException(DefaultExecuteContext.java:755) ~[jooq-3.12.1.jar:?]
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:383) ~[jooq-3.12.1.jar:?]
	at org.jooq.impl.AbstractResultQuery.fetch(AbstractResultQuery.java:353) ~[jooq-3.12.1.jar:?]
	at org.jooq.impl.SelectImpl.fetch(SelectImpl.java:2693) ~[jooq-3.12.1.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppTagRepositoryImpl.getRecords(OcppTagRepositoryImpl.java:128) ~[steve.jar:?]
	at de.rwth.idsg.steve.service.OcppTagServiceImpl.getAuthData(OcppTagServiceImpl.java:65) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.task.SendLocalListTask.createOcpp16Request(SendLocalListTask.java:103) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.task.SendLocalListTask.<init>(SendLocalListTask.java:43) ~[steve.jar:?]
	at de.rwth.idsg.steve.service.ChargePointService15_Client.sendLocalList(ChargePointService15_Client.java:113) ~[steve.jar:?]
	at de.rwth.idsg.steve.web.controller.Ocpp15Controller.postSendList(Ocpp15Controller.java:220) ~[steve.jar:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:190) ~[spring-web-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:138) ~[spring-web-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:104) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:892) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:797) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1039) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:942) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1005) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:908) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:707) ~[javax.servlet-api-3.1.0.jar:3.1.0]
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:882) ~[spring-webmvc-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790) ~[javax.servlet-api-3.1.0.jar:3.1.0]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:876) ~[jetty-servlet-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1623) ~[jetty-servlet-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:320) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke(FilterSecurityInterceptor.java:127) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter(FilterSecurityInterceptor.java:91) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:119) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:137) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter(AnonymousAuthenticationFilter.java:111) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter(SecurityContextHolderAwareRequestFilter.java:170) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.savedrequest.RequestCacheAwareFilter.doFilter(RequestCacheAwareFilter.java:63) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.authentication.www.BasicAuthenticationFilter.doFilterInternal(BasicAuthenticationFilter.java:158) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:118) ~[spring-web-5.1.9.RELEASE.jar:5.1.9.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.authentication.AbstractAuthenticationProcessingFilter.doFilter(AbstractAuthenticationProcessingFilter.java:200) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.authentication.logout.LogoutFilter.doFilter(LogoutFilter.java:116) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
	at org.springframework.security.web.csrf.CsrfFilter.doFilterInternal(CsrfFilter.java:124) ~[spring-security-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
[... cut ...]
	at org.jooq.tools.jdbc.DefaultPreparedStatement.execute(DefaultPreparedStatement.java:209) ~[jooq-3.12.1.jar:?]
	at org.jooq.impl.Tools.executeStatementAndGetFirstResultSet(Tools.java:3928) ~[jooq-3.12.1.jar:?]
	at org.jooq.impl.AbstractResultQuery.execute(AbstractResultQuery.java:294) ~[jooq-3.12.1.jar:?]
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:369) ~[jooq-3.12.1.jar:?]
	... 95 more

----------------------------------------------------------
Note
-----
I have implemented the solution suggested in the issue # 215 but still having the same problem.

goekay, 2020-06-01:
> I have implemented the solution suggested in the issue # 215 but still having the same problem.

what do you mean by that? if you use a release after this bugfix, it'll include the solution.

ShadElbo, 2020-06-01:
> > I have implemented the solution suggested in the issue # 215 but still having the same problem.
> 
> what do you mean by that? if you use a release after this bugfix, it'll include the solution. you don't have to do anything.

There a suggestion in the  the issue # 215 to modify a line of code which I did. However, I'm still getting the same error.

goekay, 2020-06-03:
sorry, but cannot reproduce the error with the latest steve release and a clean database.

moreover, i still cannot understand why you modified the code, even though i fixed it. since i don't know what you changed and how you did it, i assume something must have gone wrong while doing it.

ShadElbo, 2020-06-05:
> sorry, but cannot reproduce the error with the latest steve release and a clean database.
> 
> moreover, i still cannot understand why you modified the code, even though i fixed it. since i don't know what you changed and how you did it, i assume something must have gone wrong while doing it.

Thanks sir for your reply, please have a look to the attached screenshot.

[image]

goekay, 2020-06-06:
i cannot help you in a meaningful manner without seeing the big picture. i am not sure why you are sending me the screenshot of my own change, commit. i am aware of that. there are other factors: did you recompile and repackage the project after that? is your database healthy and has the latest changes? and so on.

we are getting into _customer support_ territory, which github issues are not. i am closing this issue. please be mindful to only open an issue if it is indeed a steve problem.

closed 2020-06-06 by goekay

ShadElbo, 2020-06-11:
> i cannot help you in a meaningful manner without seeing the big picture. i am not sure why you are sending me the screenshot of my own change, commit. i am aware of that. there are other factors: did you recompile and repackage the project after your modification? is your database healthy and has the latest changes? and so on.
> 
> we are getting into _customer support_ territory, which github issues are not. i am closing this issue. please be mindful to only open an issue if it is indeed a steve problem.

I think it would be appreciated if your reply was more respectful than this. You can ignore my issue but do not answer like that.

goekay, 2020-06-11:
sorry for the situation, but i have a list of checkpoints to gauge how much time i should invest into an issue and how much attention your problem deserves:

first of all, you bypassed the template and opened an issue without the proper format. you willingly, by choice, actively stepped over the templates page. this is a direct red flag for me. _you_ disrespect me, because it causes more effort on my part to help you. please read again: to help you. and i am doing this without any monetary incentives since 2013. this is not my job.

there is no group of people behind steve. there is no company. it is a pretty small operation. we cannot deal with every possible simple problem of users related to installation, database preparation, etc. this is why i said that we cannot afford to provide any customer support (which is the common business model of may open source projects). there is no "customer care" people answering mails and telephone calls. the issue should really really really be about a problem of steve. 

now... let's go on with the analysis. secondly, you manually change some code (even though i fixed it in a previous release/version. which version are you using? i don't know, because you did not use the issue template. you get my point?). why did you do it -- i don't know. what is your experience level with java projects -- i don't know. a complete black box for me. thirdly, despite the red flags so far, i even tried to reproduce the error, but could not, which i reported to you. i still don't know what the cause of your issue is. 

but you did not like the tone of my answer, even though, if you look at the facts, it is pretty clear that you were disrespectful towards me. do not post comments like that.

ShadElbo, 2020-06-11:
Sorry, I'm not aware of the template. I apologize for my comment (I have deleted it).