steve-community/steve #360: Create SteVe Instance in AWS Lightsail. Stuck in communication

PBrockhoff, 2020-06-13:
I am trying to follow the instructions for the Steve OCPP server in AWS lightsail
https://github.com/RWTH-i5-IDSG/steve/wiki/Create-SteVe-Instance-in-AWS-Lightsail
I did try the steps several times with the same outcome
Unable to obtain connection from database (jdbc:mysql://3.126.93.141:3306/stevedb?useSSL=true&serverTimezone=UTC) for user 'dbmasteruser': Communications link failur
e

Is there a solution for this. 

Flyway Community Edition 6.4.0 by Redgate
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:17 min
[INFO] Finished at: 2020-06-13T20:14:32Z
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:6.4.0:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: 
[ERROR] Unable to obtain connection from database (jdbc:mysql://3.126.93.141:3306/stevedb?useSSL=true&serverTimezone=UTC) for user 'dbmasteruser': Communications link failur
e
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
[ERROR] ---------------------------------------------------------------------------------------------------------------------------------------------------------------------
-----------------------------------------------------------------------------------------------------------------------------------
[ERROR] SQL State  : 08S01
[ERROR] Error Code : 0
[ERROR] Message    : Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
[ERROR] :

goekay, 2020-06-14:
@csamsel ?

csamsel, 2020-06-14:
You have to put the MySQL endpoint hostname into the db.ip field, not your public IP.

PBrockhoff, 2020-06-14:
Thanks it works, great stuff

closed 2020-06-14 by PBrockhoff