steve-community/steve #367: ChargingProfilePurpose missing parameter for TxProfile

whichnick, 2020-07-08:
In a request from SetChargingProfile.req, when the field is set to chargeProfilePurpose TxProfile, we should have the transactionId parameter.
regarding the specification Open Charge Point Protocol 1.6, edition 2 FINAL, 2017-09-28, we can see that this parameter is optional, but we must define it when a transaction is already in progress

[2,"8effebba-c4d6-4b43-96a1-eb16a7be3d9b","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute",
"chargingSchedule":{"duration":160,"startSchedule":"2020-07-02T12:01:00.000Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":8,"numberPhases":1},{"startPeriod":50,"limit":11,"numberPhases":1},{"startPeriod":140,"limit":16,"numberPhases":1},{"startPeriod":200,"limit":6,"numberPhases":1},{"startPeriod":240,"limit":12,"numberPhases":1}],"minChargingRate":10}}}]

[4, "8effebba-c4d6-4b43-96a1-eb16a7be3d9b", "PropertyConstraintViolation", "Validation failed: Transaction Id is empty for TxProfile", {}]

goekay, 2020-07-09:
sorry, this was indeed a mistake. but the mistake is not that transaction id cannot be set for TxProfile. 

TxProfile should not be there at all (in the possible values for selection), because in the current charging profile implementation we do not support setting it for a transaction. since this is a quite specific case, i could not figure out what the process and flow should be. and i had a lot of open questions. therefore, i left it out. maybe we can use this issue to talk about it. 

one of the things i was not sure about is the list of sanity checks we might do. is this overengineering:
- transaction should be at the same connector id as the charging profile definition
- transaction should be ongoing (no transaction-stop yet)
- transaction's time period should overlap with transaction (startSchedule does not necessarily have to be before or after transaction-start)

additional question is whether we should store this in a database table. current db schema allows to assign a profile to a connector. we probably need another table for profile->transaction mappings. but.. is this too much complexity? maybe we need to define a charging profile in a fire-and-forget fashion, since transactions a short-lived. in this case, current profile management flow is not fitting.

the list goes on and on :)

lategoodbye, 2020-07-12:
> one of the things i was not sure about is the list of sanity checks we might do. is this overengineering:
> 
>     * transaction should be at the same connector id as the charging profile definition

Maybe we should look at the user perspective. The charging profile purpose defines different use cases (transaction related or not). So it might be better to select this at first. Based on this we can filter for the suitable charge profile IDs. In case of ChargePointMaxProfile and TxDefaultProfile the interface offer the connector ID as an input field (as before). But in case of TxProfile only the ongoing transactions ID as a select field. So Steve could retrieve the connector ID from the transaction data.

>     * transaction should be ongoing (no transaction-stop yet)

OCPP 1.6 has a hint about this:

_It  is  RECOMMENDED  to  check  the  timestamp  in  the StartTransaction.req  PDU  prior  to  sending  a charging profile to check if the transaction is likely to be still ongoing. The StartTransaction.req might have been cached during an offline period._

>     * transaction's time period should overlap with transaction (startSchedule does not necessarily have to be before or after transaction-start)
> 

Personally i would leave such checks out.

> or should steve allow any input and hope that the charging station handles these cases?

gianmarcoodorizzi, 2021-12-30:
Hi @goekay,
I am using Steve as a test Central System to validate our Charge Point software and I need to send a TxProfile. Would it be possible to simply add a transactionId field for charging profiles? As @whichnick pointed out, that field is optional, but required when TxProfile is used.
Currently I am unable to test TxProfiles...

lategoodbye, 2022-11-05:
@gianmarcoodorizzi Please take a look at the hack above.

gianmarcoodorizzi, 2022-11-05:
> 

Thanks for this!

lategoodbye, 2025-09-13:
@goekay 
I'm struggling to port even at the first patch. In the old branch I extended SetChargingProfileParams with a transactionId and checked the constraints within ChargePointService16_Client.java. Now this file doesn't exist anymore and the checks are done in SetChargingProfileTask.java. But SetChargingProfileParams and the member transactionId isn't available checkAdditionalConstraints.

How can I solve this?

goekay, 2025-09-14:
okay, let me do this. i will cherry pick your 3 commits while fixing the commits in feature branch. PR will be mine, but at least the commits will be yours ;)

lategoodbye, 2025-09-14:
Thanks

I think you can ignore https://github.com/chargebyte/steve/commit/15de0ac48a93d483e074ebe618f4ea151f877d9a because it's already merged.

I rewrote the commit message for patch 1:

````
Add input for TxProfile transaction id #367

According to OCPP 1.6 spec a SetChargingProfile.req with 
chargeProfilePurpose TxProfile should have a transactionId parameter.
This small hack adds a simple text input field and some simple checks
for the transaction id.
````