steve-community/steve #369: [Question] How to connect to SteVe via websocat client?

connieyang, 2020-07-14:
I built and deployed SteVe via the provided docker on my AWS test account.  I was able to access the web interface from web browser using http://<my-test-AWS-ip-address>:8180 with default login and etc.  Following this step, I registered a test charge point with an id, 8888, using the browser.

Now, when I tried to access the charge point information using a websocket CLI client like

```
websocat ws://<my-test-AWS-ip-address>:8180/steve/websocket/CentralSystemService/8888`
```
 
I got the following error
```
websocat: WebSocketError: Received unexpected status code
websocat: error running
```

Do you know why?

goekay, 2020-07-14:
hello, such a plain websocket connection attempt will not be granted by steve, because ocpp requires a subprotocol to be set. its value is the ocpp version. this helps a backend to identify the version of the connecting station.

connieyang, 2020-07-14:
Thanks.  Figured it out.

closed 2020-07-14 by connieyang