steve-community/steve #373: OCPP 2.0.1?

retrocoder-78, 2020-07-21:
Hello steve team!

The work done on this software is impressive. I am curious if their has been any movement (or conversation) towards the OCPP 2.0.1? Or, what the challenges might be in doing so. The last time this was mentioned looks to be 2018'ish. So, just curious on an update if one exists.

Thank you for your time!

goekay, 2020-07-22:
hey, 

thanks for the kind words. i am for transparency and there would be some trace on github, if there is any kind of development. unfortunately, i still cannot find time to look into new ocpp versions. therefore, i cannot say what the challenges would be.

retrocoder-78, 2020-07-23:
Thank you for the quick update goekay! I'll mark this as closed.

Cheers!

closed 2020-07-23 by retrocoder-78