steve-community/steve #375: [Help] Debugging Application from IntelliJ fails to resolve any views

ashevade1, 2020-07-23:
These are the steps performed
1) Imported SteVe into IntelliJ. It builds wtihout issues.
2) Start a debug session by selecting chosing the Application.main() 
3) The application starts fine. 

An attempt to access any UI page results into a 404 not found error. 
For e.g 
  http://localhost:8080/steve/manager/home
Response:
HTTP ERROR 404
Problem accessing /steve/WEB-INF/views/home.jsp. Reason:

    /steve/WEB-INF/views/home.jsp

Please refer to the screen shot attached.

4) The same URL works when the application is started from the command line
for e.g if we start it with 
 java -jar ./target/steve.jar  

The same URL leads to a valid page being shown.

5) It does appear there is some issue with resolving views/ Jsps from within IntelliJ

Can someone please help? Any pointers are appreciated.

Best Regards
Ani
[image]

kokhanchyk, 2020-09-02:
@ashevade1 
sometimes also getting the same error. 
to fix it, I stop application. 
then make mvn clean
then mvn package or install
and then run from idea and it runs.

goekay, 2020-09-02:
jsp files need to be compiled during the compilation phase. i guess you might see this error, if they were not.

i suspect that the regular intellij build configuration does not include that. i always remove the normal build, and add `mvn compile` as the prelaunch step. if you want to be extra safe you can use `mvn clean compile` with increased compile times.

closed 2020-12-11 by goekay