steve-community/steve #376: Liteon 1.6J OCPP compliant evse not connecting....

retrocoder-78, 2020-07-24:
Hello!

Wanting to play and get familiar with the latest SteVe software, I am attempting to connect an evse I have in my possession. I am using a raspberry pi as the server. I have followed the SteVe setup and everything works when I test within my home network. 

https://github.com/victormunoz/OCPP-1.6-Chargebox-Simulator

That's to say that the html testing does what it should do within the SteVe software. Which is really great. 

But, when I point the evse (defaulted to factory settings) at the same ws address as the test html but with the correct chargebox id registered in the test html and SteVe, nothing. I have changed options and setup SteVe a couple of times and the same result. I have changed OCPP options in the firmware of the charger too. There must be something simple I am missing....I have played with a couple of different evse's in regards to OCPP and they range from well laid out in their firmware to a total mess. This liteon unit seems pretty well laid out in regards to OCPP. So, seemingly should work with SteVe with little effort. But, not yet....

Any thoughts, comments, or a point in the correct direction would be much appreciated.

I have included some screenshots and the manual of the evse as support documentation.

Thanks!

-----------------
[Users-Manual-rev-pdf-4325734.pdf](https://github.com/RWTH-i5-IDSG/steve/files/4973830/Users-Manual-rev-pdf-4325734.pdf)
-----------------
[image]
[image]
[image]

goekay, 2020-07-25:
hey, i don't think you are supposed to add the charge point id to the central system url as it is the case in your screenshot. yes, it is required to be there, but i think charging station software might add it automatically.

i imagine that the url in your case contains the id twice, e.g. `ws://.../1A30.../1A30`

retrocoder-78, 2020-07-25:
> hey, i don't think you are supposed to add the charge point id to the central system url as it is the case in your screenshot. yes, it is required to be there, but i think charging station software might add it automatically.
> 
> i imagine that the url, which the station will use to connect, in your case contains the id twice, e.g. `ws://.../1A30.../1A30...`

Thanks for taking a look. It might be that simple. I'll make the change and report back. 

Thanks!

retrocoder-78, 2020-07-25:
Yup, that was the problem. Heartbeat received and it's all connected through SteVe. Thanks goekay for taking a look. Really appreciate your time!

closed 2020-07-25 by retrocoder-78