steve-community/steve #385: Play Framework integration

cyrilfr, 2020-08-10:
I would like to use SteVe in my Play Framework 2.8 app. The Play Framework uses Akka as a Web server with its own [WebSocket](https://www.playframework.com/documentation/2.8.x/JavaWebSockets) implementation.

How should I proceed to integrate SteVe in my app?

goekay, 2020-08-13:
i am aware of play and akka, but am not familiar with them in detail afaik, play is a framework much like spring. since steve uses spring, i don't think it will be easy to integrate it into your app. basically, you would have to strip away all framework layers and jetty etc. such that only basic functionality remains which you can use in your app. very error-prone and not worth it imo.

goekay, 2020-09-27:
i consider this question answered and therefore close it.

closed 2020-09-27 by goekay