steve-community/steve #387: How to change login / password after compilation?

DeltaVetal26, 2020-08-15:
Hi,

Before compiling Steve, it was possible to specify settings in the configuration file, including the username and password.

As far as I understood, this only works before compilation.
Now I have a task to change the login and password.
I found the user table in MySQL, but there was no data there.

Questions:
1) How can I change the password in the compiled version?
2) Is it possible to create multiple accounts to access Steve panel?

Thank you!

csamsel, 2020-08-15:
1. To set the the login username and password you have to rerun maven - correct.
2. No

DeltaVetal26, 2020-08-16:
@csamsel ,

Thank you for the answer! Ok.

closed 2020-08-16 by DeltaVetal26

goekay, 2020-08-16:
> I found the "user" table in MySQL, but there was no data there.

i want to clarify that this user table has nothing to do with the user of steve (for ex. charging station manager) who has to login to use the web interface. the definition of this user is the holder of an rfid card (ocpp tag).

dulaimingming, 2020-09-22:
Hello, why can't I connect to STEVE, and what does the authkey mean?

DeltaVetal26, 2020-09-24:
Hello @dulaimingming,

If you are talking about Steve configuration (main.properties file), then there is no such thing. AuthKey - authorization key. Where exactly did you find it?

dulaimingming, 2020-09-24:
I can connect!  Thank you for your reply