steve-community/steve #388: Implement REST Endpoint for Start and Stop Transaction

Safari94, 2020-08-17:
Hi,

I am trying to implement two REST endpoints for start and stop a charging transaction.
Based on the code already done, I am using the class `ChargePointService12_Client` for doing so.

Basically I am only getting JSON on the body and convert it to both `RemoteStartTransactionParams` and `RemoteStopTransactionParams`

But, once I try to pass command and run it, I am having the following error.
[image]

My code:
`

@RestController
@RequestMapping("/api")
public class APIConnector {

    private static final String TEST = "test";

    private static final String START_TRANSACTION="startTransaction";

    private static final String STOP_TRANSACTION="stopTransaction";

    @Autowired
    @Qualifier("ChargePointService12_Client")
    private ChargePointService12_Client client12;

    @Autowired private TransactionRepository transactionRepository;

    protected ChargePointService12_Client getClient12() {
        return client12;
    }

    @RequestMapping(value = TEST, method = RequestMethod.GET)
    public String testEndpoint(){
        return "sucess";
    }

    @RequestMapping(value = START_TRANSACTION, method = RequestMethod.POST)
    public Map<String, String> startTransaction(@RequestBody Map request){

        HashMap<String, String> res = new HashMap<>();

        //List<ChargePointSelect> chargePointSelectList=this.chargePointRepository.getChargePointSelectfromID(request.get("stationID").toString());
        //if(chargePointSelectList.size()==0){

        ChargePointSelect chargePointSelect=new ChargePointSelect(OcppTransport.JSON,"HELLO","-");
        List<ChargePointSelect> chargePointSelectList=new ArrayList<>();
        chargePointSelectList.add(chargePointSelect);
        RemoteStartTransactionParams remoteStartTransactionParams = new RemoteStartTransactionParams();
        remoteStartTransactionParams.setChargePointSelectList(chargePointSelectList);
        remoteStartTransactionParams.setConnectorId(Integer.parseInt(request.get("connectorID").toString()));
        remoteStartTransactionParams.setIdTag(request.get("tagID").toString());

        int result=this.client12.remoteStartTransaction(remoteStartTransactionParams);
        res.put("result",String.valueOf(result));
        return res;
    }

    @RequestMapping(value = STOP_TRANSACTION, method = RequestMethod.POST)
    public Map<String, String> stopTransaction(@RequestBody Map request){

        HashMap<String, String> res = new HashMap<>();

        List<Integer> ids=transactionRepository.getActiveTransactionIds(request.get("stationId").toString());
        if (ids.isEmpty()){
            res.put("result","nok");
            res.put("message","Transactions not found");
            return res;
        }
        ChargePointSelect chargePointSelect=new ChargePointSelect(OcppTransport.JSON,request.get("stationId").toString(),"-");
        List<ChargePointSelect> chargePointSelectList=new ArrayList<>();
        chargePointSelectList.add(chargePointSelect);
        RemoteStopTransactionParams remoteStopTransactionParams= new RemoteStopTransactionParams();
        remoteStopTransactionParams.setTransactionId(ids.get(0));
        remoteStopTransactionParams.setChargePointSelectList(chargePointSelectList);

        int result=this.client12.remoteStopTransaction(remoteStopTransactionParams);

        res.put("result",String.valueOf(result));
        return res;

    }

}
`

Can anyone help me?

Thanks in advance

redhell, 2020-08-17:
You may want to checkout my approach https://github.com/redhell/steve/tree/REST_Branch

Safari94, 2020-09-01:
Hi,

First of all many thanks @redhell 
I am facing an issue when using your code.
I am trying to start to sessions, using 2 different occp tag's in the same charger. (I am able to start both without any problem)
I am able to stop one session but when trying to close the other one I am getting HTTP status code 409 (conflit)

Do you have any idea what can be causing this issue?

Once again thanks

redhell, 2020-09-01:
@Safari94 Maybe it is related to the connector outlet. I assume you want to start a charging session on charger with multiple outlets. 
I wrote the code with a charging station with a single outlet. 

To fix the problem the you have to get the outlet number (1/2/3) and pass it to the /startSession/
(-> params.setConnectorId(0);)
Should be the same for stopSession

Safari94, 2020-09-02:
Hi @redhell,

In the startSession I changed the code and I am passing the connectorID as a requestParam.
Like I told you before, I am able to open a session in 2 different outlets, using 2 different occp tags

The problem is when I am trying to stop session, I am able to close one of them but when trying to close the other one I get 409 conflit.

I review your code and in the stopSession there isn't any reference to the connector ID. As far as I understood, you are checking for that charge box how many transactions are active and by using the occp tag you close the one that belongs to that ocpp tag, right?

Do you have any idea about what can be the problem?

Thanks.

redhell, 2020-09-04:
As far as I can remember the remoteStoptransaction stops the last active transaction. 
You'll get the conflict if no active transaction is stored ( if (transactionIDs.size() > 0)). 

In OCPP you cannot send the  connectorID via a remoteStopTransaction.

An approach for you: When starting a charging session store the transaction ID with the chargebox id + outlet. 
OR: Override the getActiveTransactionIds function (probably the best approach)
    @Override
    public List<Integer> getActiveTransactionIds(String chargeBoxId, int connectorID) {
        return ctx.select(TRANSACTION.TRANSACTION_PK)
                  .from(TRANSACTION)
                  .join(CONNECTOR)
                    .on(TRANSACTION.CONNECTOR_PK.equal(CONNECTOR.CONNECTOR_PK))
                    .and(CONNECTOR.CHARGE_BOX_ID.equal(chargeBoxId))
                    .and(CONNECTOR.CHARGE_BOX_ID.equal(connectorID))
                  .where(TRANSACTION.STOP_TIMESTAMP.isNull())
                  .fetch(TRANSACTION.TRANSACTION_PK);
    }

Shivraj-Patil, 2021-03-05:
Hi @redhell,
Can you please tell me what are the endpoints (URLs) for remotely calling getting status, start transaction and stop transactions? And do I need to do any configuration to use them?

Shivraj-Patil, 2021-03-05:
Hi @redhell,
Can you please give me some examples of how these endpoint URLs could be called using few examples?

Shivraj-Patil, 2021-03-05:
Hi @redhell,
I am able to use your code to get connectors info & get chargers configurations, but I'm not able to start/stop a session. My charge points use OCPP 1.5S.

andresxz32, 2022-02-17:
@Shivraj-Patil Hi, were you able to solve it? I have the same problem

redhell, 2022-02-17:
It was only implemented for OCPP 1.6 Endpoints

theGurk, 2022-03-02:
@redhell have you any idea how to configure the SecurityConfiguration (\src\main\java\de\rwth\idsg\steve\config\SecurityConfiguration) if you would like to authorize the the REST endpoint too? I have made something similar like you with a REST-controller but have not been able to figure out how to configure the authorization if I would like to authorize the admin user that we set in the main.properties file.

Do you have any idea?

rahulbhat13, 2022-04-07:
Is the code been updated and upgraded to support multiple chargers ? Can we have the readme to support the steps

Safari94, 2022-05-18:
hi @rahulbhat13 

There you have the code that i am using to activate multiple connectors in the same station

`  @GetMapping(value = sCHARGEBOXID + "/startSession/{idTag}")
    public void startRemoteSession(@PathVariable("chargeBoxId") String chargeBoxId,
                                   @PathVariable("idTag") String idTag,@RequestParam("connectorID") int connectorID,
                                   HttpServletResponse response) throws IOException {
        try {
            if (!getTokenList(idTag).isEmpty()) {
                RemoteStartTransactionParams params = new RemoteStartTransactionParams();
                params.setIdTag(idTag);
                params.setConnectorId(connectorID);
                List<ChargePointSelect> cp = new ArrayList<>();
                ChargePointSelect cps = new ChargePointSelect(OcppTransport.JSON, chargeBoxId);
                cp.add(cps);
                params.setChargePointSelectList(cp);
                CommunicationTask task = taskStore.get(client16.remoteStartTransaction(params));

                while (!task.isFinished() || task.getResultMap().size() > 1) {
                    System.out.println("wait for");
                }
                RequestResult result = (RequestResult) task.getResultMap().get(chargeBoxId);
                if (result.getResponse() == null) {
                    response.setStatus(HttpServletResponse.SC_PRECONDITION_FAILED);
                } else if (!result.getResponse().equals("Accepted")) {
                    response.setStatus(HttpServletResponse.SC_FORBIDDEN);
                    writeOutput(response, objectMapper.writeValueAsString(result.getResponse()));
                } else {
                    writeOutput(response, objectMapper.writeValueAsString(result.getResponse()));
                }
            }
        } catch (NullPointerException nullPointerException) {
            response.setStatus(HttpServletResponse.SC_FORBIDDEN);
        }
    }`

rahulbhat13, 2022-08-07:
@theGurk Were you able to solve the issue , since I am facing the issue to setup the rest endpoint with the authorisation since not able to understand what kind of mechanism was used  and exactly where the changes are done in the code of @redhell

juherr, 2022-10-30:
Related to #910
When the API will be provided, it will be easy to add endpoints for start/stop transactions (or any other operations).

goekay, 2024-02-18:
closing this since we have the meta ticket.

closed 2024-02-18 by goekay

Mehmedx7, 2024-07-14:
> You may want to checkout my approach https://github.com/redhell/steve/tree/REST_Branch

cloned your code but getting this erorr:
/bin/sh: 1: ./mvnw: not found