steve-community/steve #391: Charge station connector appearing duplicated with multiple statuses

Borduhh, 2020-09-01:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.3
Operating system  : Ubuntu 18.04.5 LTS
JDK               :  11.0.8
Database          :  MySQL 5.7.30
```

#### Expected Behavior
`/steve/manager/home/connectorStatus` table shows the correct amount of charging ports with their correct status.

#### Actual Behavior
When `StopTransaction` occurs, `/steve/manager/home/connectorStatus` duplicates the connector ports, showing one with the status `Finishing` and one with the status of `Available`.

#### Steps to Reproduce the Problem

  1. Register a charging station ID into SteVe
  2. Connect the charging station port to the EV and disconnect it after few minutes
  3. Observe the status of the charging station on `/steve/manager/home/connectorStatus`

#### Additional context

We checked the logs and can see both `StatusNotifications`, one for `Finishing` and one for `Available`

```
[INFO ] 2020-09-01 18:51:00,203 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=xxx, sessionId=8c67579d-1942-27e7-63ed-35aec5e1a951] Received: [2,"1598986260136","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Finishing","timestamp":"2020-09-01T18:50:56.000Z","vendorErrorCode":"PLUGGED-IN"}]
[INFO ] 2020-09-01 18:51:00,215 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=xxx, sessionId=8c67579d-1942-27e7-63ed-35aec5e1a951] Sending: [3,"1598986260136",{}]

[INFO ] 2020-09-01 18:51:02,193 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=xxx, sessionId=8c67579d-1942-27e7-63ed-35aec5e1a951] Received: [2,"1598986262232","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2020-09-01T18:50:56.000Z","vendorErrorCode":"NoError"}]
[INFO ] 2020-09-01 18:51:02,204 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=xxx, sessionId=8c67579d-1942-27e7-63ed-35aec5e1a951] Sending: [3,"1598986262232",{}]
```

[image]

...

goekay, 2020-09-01:
hey, thanks for the detailed report.

it's unfortunate that this is happening. as you can see, the timestamps of both events are the same up to the second. the logic for this table selects the status for the latest timestamp. and in your example, two status values exist. i don't know what a better strategy would be. we, for sure, could select one and mask the other but this wouls hide information from the user, which might have unwanted implications.

Borduhh, 2020-09-01:
@goekay It looks like in the logs that the notifications are milliseconds apart. `2020-09-01 18:51:00` for the `finishing` and `2020-09-01 18:51:02` for the `available`. Maybe we can bring the time calculation down to milliseconds so the logic would work right?

goekay, 2020-09-01:
we cannot trust the timestamp of the log entry.

example: if station became offline for a while, and reported status values for the time in between after coming online later.

Borduhh, 2020-09-01:
Wouldn't it eventually get to the right status though even with the log values? I am not 100% positive, but I believe that they are designed to cache messages in order, no?

goekay, 2020-09-01:
what i am saying is, the order of the status event messages arriving do not have to be the same as the events happening.

Borduhh, 2020-09-01:
Agreed. Would calculating in milliseconds though solve this problem for most of the use cases?

goekay, 2020-09-01:
sure, the higher the resolution of the timestamp (the timetamp field within the message, not the log entry timestamp) the lower the propability of two things happening at the same time at this connector :)

can you configure your charging station to use more precise timestamps?

Borduhh, 2020-09-01:
Ah! This is the whole 

> I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

part. Thanks for the help! I'll close the issue,

closed 2020-09-01 by Borduhh