steve-community/steve #393: There is no communication with the JSON 1.6 Setec charger

Kaniiisek, 2020-09-02:
Hello. I'm trying to connect my Setec 20kW charger with Steve version 3.0.2 to a Raspberry for the first time.
If I enter the destination address with the charger ID into the charger, then in my case:
ws: //10.109.0.XXX: 8080 / steve / websocket / CentralSystemService / CBI-15xxxxxxxxx
Should I automatically contact Steve on raspberry and start communicating?
I tried Simple OCPP 1.6 Chargebox Simulator and it went fine.
Is there something I'm doing wrong?

goekay, 2020-09-02:
hi, please look at a similar issue: https://github.com/RWTH-i5-IDSG/steve/issues/376

Kaniiisek, 2020-09-03:
YES !!! It helped. Thank you very much. I'm really happy now.

closed 2020-09-03 by goekay

csamsel, 2020-09-06:
@goekay seems to be common pitfall, maybe we should slightly rephrase the field name in the GUI?

goekay, 2020-09-06:
@csamsel i am not sure what part of the GUI you mean, but i think adjusting readme might be enough: https://github.com/RWTH-i5-IDSG/steve#add-a-charge-point