steve-community/steve #395: Error with Clock Aligned Meter Values

Borduhh, 2020-09-08:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Expected Behavior
STEVE saves the clock aligned meter value successfully.

#### Actual Behavior
```
[ERROR] 2020-09-08 19:45:06,250 de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl - Exception occurred
org.jooq.exception.DataAccessException: SQL [insert into `stevedb`.`connector_meter_value` (`connector_pk`, `transaction_pk`, `value_timestamp`, `value`, `reading_context`, `format`, `measurand`, `location`, `unit`, `phase`) values (?, ?, ?, ?, ?, ?, ?, ?, ?, ?)]; Cannot add or update a child row: a foreign key constraint fails (`stevedb`.`connector_meter_value`, CONSTRAINT `FK_tid_cm` FOREIGN KEY (`transaction_pk`) REFERENCES `transaction_start` (`transaction_pk`) ON DELETE SET NULL ON UPDATE NO ACTION)
	at org.jooq_3.12.4.MYSQL.debug(Unknown Source) ~[?:?]
	at org.jooq.impl.Tools.translate(Tools.java:2728) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.DefaultExecuteContext.sqlException(DefaultExecuteContext.java:755) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.BatchSingle.executePrepared(BatchSingle.java:253) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.BatchSingle.execute(BatchSingle.java:176) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.BatchCRUD.executePrepared(BatchCRUD.java:155) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.BatchCRUD.execute(BatchCRUD.java:96) ~[jooq-3.12.4.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.batchInsertMeterValues(OcppServerRepositoryImpl.java:442) ~[steve.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.lambda$insertMeterValues$1(OcppServerRepositoryImpl.java:171) ~[steve.jar:?]
	at org.jooq.impl.DefaultDSLContext$3.run(DefaultDSLContext.java:608) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.DefaultDSLContext$3.run(DefaultDSLContext.java:605) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.DefaultDSLContext.lambda$transactionResult0$0(DefaultDSLContext.java:536) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.Tools$12$1.block(Tools.java:4853) ~[jooq-3.12.4.jar:?]
	at java.util.concurrent.ForkJoinPool.managedBlock(ForkJoinPool.java:3128) ~[?:?]
	at org.jooq.impl.Tools$12.get(Tools.java:4850) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.DefaultDSLContext.transactionResult0(DefaultDSLContext.java:588) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.DefaultDSLContext.transactionResult(DefaultDSLContext.java:505) ~[jooq-3.12.4.jar:?]
	at org.jooq.impl.DefaultDSLContext.transaction(DefaultDSLContext.java:605) ~[jooq-3.12.4.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.insertMeterValues(OcppServerRepositoryImpl.java:165) ~[steve.jar:?]
	at de.rwth.idsg.steve.service.CentralSystemService16_Service.meterValues(CentralSystemService16_Service.java:149) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.CentralSystemService16_SoapServer.meterValues(CentralSystemService16_SoapServer.java:101) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint$Ocpp16CallHandler.dispatch(Ocpp16WebSocketEndpoint.java:88) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.AbstractCallHandler.accept(AbstractCallHandler.java:46) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.IncomingPipeline.accept(IncomingPipeline.java:61) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleTextMessage(AbstractWebSocketEndpoint.java:107) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.onMessage(AbstractWebSocketEndpoint.java:79) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ConcurrentWebSocketHandler.handleMessage(ConcurrentWebSocketHandler.java:50) ~[steve.jar:?]
	at org.springframework.web.socket.adapter.jetty.JettyWebSocketHandlerAdapter.onWebSocketText(JettyWebSocketHandlerAdapter.java:83) ~[spring-websocket-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at jdk.internal.reflect.GeneratedMethodAccessor63.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:71) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:241) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:65) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:233) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:150) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:319) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:201) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:226) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.Parser.parseSingleFrame(Parser.java:260) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.readParse(AbstractWebSocketConnection.java:583) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:441) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:428) ~[websocket-common-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117) ~[jetty-io-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:781) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:917) ~[jetty-util-9.4.19.v20190610.jar:9.4.19.v20190610]
	at java.lang.Thread.run(Thread.java:834) [?:?]
[... 22 lines cut ...]
	at com.mysql.cj.jdbc.ClientPreparedStatement.executeUpdateInternal(ClientPreparedStatement.java:1092) ~[mysql-connector-java-8.0.21.jar:8.0.21]
	at com.mysql.cj.jdbc.ClientPreparedStatement.executeUpdateInternal(ClientPreparedStatement.java:1040) ~[mysql-connector-java-8.0.21.jar:8.0.21]
[... cut ...] .57","context":"Other","format":"Raw","measurand":"Energy.Active.Import.Register","phase":"L1-N","location":"Body","unit":"Wh"},{"value":"0.0","context":"Other","format":"Raw","measurand":"Power.Active.Import","phase":"L1-N","location":"Body","unit":"W"},{"value":"0.0","context":"Other","format":"Raw","measurand":"Current.Import","phase":"L1-N","location":"Body","unit":"A"},{"value":"0.0","context":"Other","format":"Raw","measurand":"Voltage","phase":"L1-N","location":"Body","unit":"V"}]}]}]
```

goekay, 2020-09-08:
is there a transaction with transactionId=0 in db?

i think you are trying to reference a transaction which does not exist. this is what the exception trace implies.

goekay, 2020-09-08:
transactionId field in MeterValues is optional. you don't have to set it, if the meter values are not related to a transaction.

goekay, 2020-12-11:
i consider the issue resolved and close it due to inactivity.