steve-community/steve #397: Can't run code.

eslavko, 2020-09-11:
Hello...
I do fresh install of ubuntu (inside virtualbox) and follow steps form github how to setup.

When I try to strat with 'java -jar target/steve.jar' I got output:
Log file: /home/slavko/logs/steve.log
Starting...................... FAILED!
Please refer to the log file for details

What I do wrong? (I don't program in java and maven is new for me.)
And here is log file:

`[INFO ] 2020-09-11 16:01:58,667 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2020-09-11 16:01:58,825 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2020-09-11 16:01:58,863 de.rwth.idsg.steve.Application - Date/time zone of the application is set to UTC. Current date/time: 2020-09-11T14:01:58.847Z
[INFO ] 2020-09-11 16:01:58,927 org.eclipse.jetty.util.log - Logging initialized @2473ms to org.eclipse.jetty.util.log.Slf4jLog
[INFO ] 2020-09-11 16:01:59,351 org.eclipse.jetty.server.Server - jetty-9.4.19.v20190610; built: 2019-06-10T16:30:51.723Z; git: afcf563148970e98786327af5e07c261fda175d3; jvm 11.0.8+10-post-Ubuntu-0ubuntu118.04.1
[INFO ] 2020-09-11 16:01:59,599 org.eclipse.jetty.server.session - DefaultSessionIdManager workerName=node0
[INFO ] 2020-09-11 16:01:59,599 org.eclipse.jetty.server.session - No SessionScavenger set, using defaults
[INFO ] 2020-09-11 16:01:59,600 org.eclipse.jetty.server.session - node0 Scavenging every 660000ms
[INFO ] 2020-09-11 16:01:59,607 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring root WebApplicationContext
[INFO ] 2020-09-11 16:01:59,609 org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization started
[INFO ] 2020-09-11 16:02:01,466 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
[INFO ] 2020-09-11 16:02:02,503 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
[INFO ] 2020-09-11 16:02:07,126 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2020-09-11 16:02:08,226 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP12
[INFO ] 2020-09-11 16:02:08,272 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2012/06/}CentralSystemService from class ocpp.cs._2012._06.CentralSystemService
[INFO ] 2020-09-11 16:02:08,477 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP15
[INFO ] 2020-09-11 16:02:08,481 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2015/10/}CentralSystemService from class ocpp.cs._2015._10.CentralSystemService
[INFO ] 2020-09-11 16:02:08,663 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP16
[INFO ] 2020-09-11 16:02:08,666 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2020-09-11 16:02:08,690 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemService
[INFO ] 2020-09-11 16:02:10,433 org.hibernate.validator.internal.util.Version - HV000001: Hibernate Validator 6.1.5.Final
[INFO ] 2020-09-11 16:02:11,167 org.springframework.security.web.DefaultSecurityFilterChain - Will secure Ant [pattern='/static/**'] with []
[INFO ] 2020-09-11 16:02:11,172 org.springframework.security.web.DefaultSecurityFilterChain - Will secure Ant [pattern='/views/**'] with []
[INFO ] 2020-09-11 16:02:11,267 org.springframework.security.web.DefaultSecurityFilterChain - Will secure any request with [org.springframework.security.web.context.request.async.WebAsyncManagerIntegrationFilter@515b7335, org.springframework.security.web.context.SecurityContextPersistenceFilter@7375ebe7, org.springframework.security.web.header.HeaderWriterFilter@433d5536, org.springframework.security.web.csrf.CsrfFilter@3958e148, org.springframework.security.web.authentication.logout.LogoutFilter@6e8426aa, org.springframework.security.web.authentication.UsernamePasswordAuthenticationFilter@4b98c80c, org.springframework.security.web.authentication.www.BasicAuthenticationFilter@3c5d60db, org.springframework.security.web.savedrequest.RequestCacheAwareFilter@207358ad, org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter@4667c6d9, org.springframework.security.web.authentication.AnonymousAuthenticationFilter@102dff25, org.springframework.security.web.session.SessionManagementFilter@6a9b8f33, org.springframework.security.web.access.ExceptionTranslationFilter@77c1a7ce, org.springframework.security.web.access.intercept.FilterSecurityInterceptor@599cf2cd]
[ERROR] 2020-09-11 16:02:11,662 org.springframework.web.context.ContextLoader - Context initialization failed
de.rwth.idsg.steve.SteveException: MySQL and Java are not using the same time zone. Java offset in seconds (0) != MySQL offset in seconds (7200)
	at de.rwth.idsg.steve.utils.DateTimeUtils.checkJavaAndMySQLOffsets(DateTimeUtils.java:107) ~[steve.jar:?]
	at de.rwth.idsg.steve.config.BeanConfiguration.afterStart(BeanConfiguration.java:168) ~[steve.jar:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.springframework.context.event.ApplicationListenerMethodAdapter.doInvoke(ApplicationListenerMethodAdapter.java:305) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.event.ApplicationListenerMethodAdapter.processEvent(ApplicationListenerMethodAdapter.java:190) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.event.ApplicationListenerMethodAdapter.onApplicationEvent(ApplicationListenerMethodAdapter.java:153) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.doInvokeListener(SimpleApplicationEventMulticaster.java:172) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.invokeListener(SimpleApplicationEventMulticaster.java:165) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.event.SimpleApplicationEventMulticaster.multicastEvent(SimpleApplicationEventMulticaster.java:139) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:404) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:361) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.support.AbstractApplicationContext.finishRefresh(AbstractApplicationContext.java:898) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:554) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.web.context.ContextLoader.configureAndRefreshWebApplicationContext(ContextLoader.java:401) ~[spring-web-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.web.context.ContextLoader.initWebApplicationContext(ContextLoader.java:292) ~[spring-web-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.web.context.ContextLoaderListener.contextInitialized(ContextLoaderListener.java:103) ~[spring-web-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.eclipse.jetty.server.handler.ContextHandler.callContextInitialized(ContextHandler.java:959) ~[jetty-server-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.servlet.ServletContextHandler.callContextInitialized(ServletContextHandler.java:553) ~[jetty-servlet-9.4.19.v20190610.jar:9.4.19.v20190610]
	at org.eclipse.jetty.server.handler.ContextHandler.startContext(ContextHandler.java:924) ~[jetty-server-9.4.19.v20190610.jar:9.4.19.v20190610]
[... cut ...]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.springframework.context.event.ApplicationListenerMethodAdapter.doInvoke(ApplicationListenerMethodAdapter.java:305) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.event.ApplicationListenerMethodAdapter.processEvent(ApplicationListenerMethodAdapter.java:190) ~[spring-context-5.2.8.RELEASE.jar:5.2.8.RELEASE]
	at org.springframework.context.event.ApplicationListenerMet

goekay, 2020-09-11:
> MySQL and Java are not using the same time zone. Java offset in seconds (0) != MySQL offset in seconds (7200)

this is the relevant part of the logs. your java application (steve) and database are using different time zones. please configure mysql to use UTC. you can look at closed issues to find out how.

eslavko, 2020-09-11:
mysql> SELECT TIMEDIFF(NOW(), UTC_TIMESTAMP);
+--------------------------------+
| TIMEDIFF(NOW(), UTC_TIMESTAMP) |
+--------------------------------+
| 00:00:00                       |
+--------------------------------+
1 row in set (0.49 sec)

seems to be allready in UTC

eslavko, 2020-09-11:
...but jre environment isn't...
thanks for info.

closed 2020-09-11 by eslavko