steve-community/steve #407: Reservation Expiring

giorgio80, 2020-10-01:
Hi Goeakey, 

More a process understanding than a bug. 

Standing to ocpp 1.6J , it is possible to delete a reservation from server. 
This is clear and work perfectly. 
On other side there's also use cases in which reservation simply expire as not been used (expiry datetime passed in reservenow - current date). 

In this case, standing to ocpp charging station should notify the server with a status notification, and this is exactly what I do.  
What I noticed is that, in this case, reservation simply remain unused in the reservation list.  Basically appended in frontend.  

So:  Do per your interpretation the chargepoint should call a cancel-reservation to server? 

looks to me  ocpp.cp._2015._10.CancelReservationRequest and cp._2015._10.CancelReservationResponse  is exactly proposed for that but there's no clear documentation about even in ocpp definition. 

Differently I'm gonna make mod  to delete unused or mark them as expired

goekay, 2020-10-01:
> Do per your interpretation the chargepoint should call a cancel-reservation to server?

no. and this is not my interpretation. please refer to the ocpp specification. these cancel reservation requests are for server to chargepoint communication.

we do not delete expired reservations from the database. we treat them the same as transactions. these are valuable usage data that can be used for billing/reporting if wanted. the frontend can filter reservations, though, and show only active ones.

but you are right. this table will grow and grow. you can delete them depending on your use case.

giorgio80, 2020-10-02:
I think you right too , the ocpp standard is completely clear and implemented right . I think is more an hole in protocol . It’s not about cancelling reservations in database but on considering them active . Es . If the admin make a reservation to a chargepoint , then the chargepoint change state for whatever reason (ex fault) , the reservation still remain active in frontend . Actually ocpp show no solution about that. So thanks for your answer as usual

goekay, 2020-10-06:
> If the admin make a reservation to a chargepoint , then the chargepoint change state for whatever reason (ex fault) , the reservation still remain active in frontend . 

it's really hard to cover every case in the protocol. your example is good, but what if the station recovers from the faulty state in 10 seconds? was it worth to cancel the reservation then? should the backend re-activate the reservation? and so on... there are some open questions. 

i think ocpp is the base and common ground for operators to implement their specific use cases on.

giorgio80, 2020-10-10:
Yep, you right... not doing on mine but other do, in the ends depends case to case :D We can close the question.

closed 2020-10-10 by goekay