steve-community/steve #408: Windows 10 maven package error

qqtt27, 2020-10-01:
I made the preparations mentioned in the Configuration and Installation section. I got the following errors in section4. `mvn package`

`[ERROR] COMPILATION ERROR :
[INFO] -------------------------------------------------------------
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[56,1] cannot find symbol
  symbol:   static TRANSACTION_START
  location: class jooq.steve.db.tables.TransactionStart
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[57,1] cannot find symbol
  symbol:   static TRANSACTION_STOP
  location: class jooq.steve.db.tables.TransactionStop
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[58,1] cannot find symbol
  symbol:   static TRANSACTION_STOP_FAILED
  location: class jooq.steve.db.tables.TransactionStopFailed
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[49,1] cannot find symbol
  symbol:   static OCPP_TAG_ACTIVITY
  location: class jooq.steve.db.tables.OcppTagActivity
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/GenericRepositoryImpl.java:[36,1] cannot find symbol
  symbol:   static RESERVATION
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/GenericRepositoryImpl.java:[37,1] cannot find symbol
  symbol:   static TRANSACTION
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/GenericRepositoryImpl.java:[40,1] cannot find symbol
  symbol:   static SCHEMA_VERSION
  location: class jooq.steve.db.tables.SchemaVersion
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[51,1] cannot find symbol
  symbol:   static TRANSACTION
  location: class jooq.steve.db.tables.Transaction
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java:[52,1] cannot find symbol
  symbol:   static TRANSACTION_START
  location: class jooq.steve.db.tables.TransactionStart
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/ChargingProfileRepositoryImpl.java:[49,1] cannot find symbol
  symbol:   static CHARGING_PROFILE
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/ChargingProfileRepositoryImpl.java:[50,1] cannot find symbol
  symbol:   static CHARGING_SCHEDULE_PERIOD
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/ChargingProfileRepositoryImpl.java:[52,1] cannot find symbol
  symbol:   static CONNECTOR_CHARGING_PROFILE
  location: class jooq.steve.db.Tables
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/SettingsRepositoryImpl.java:[44,1] cannot find symbol
  symbol:   static SETTINGS
  location: class jooq.steve.db.tables.Settings
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/ReservationRepositoryImpl.java:[47,1] cannot find symbol
  symbol:   static RESERVATION
  location: class jooq.steve.db.tables.Reservation
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java:[69,22] cannot find symbol
  symbol:   method builder()
  location: class de.rwth.idsg.steve.SteveConfiguration.Jetty
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java:[80,16] cannot find symbol
  symbol:   method builder()
  location: class de.rwth.idsg.steve.SteveConfiguration.DB
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java:[91,20] cannot find symbol
  symbol:   method builder()
  location: class de.rwth.idsg.steve.SteveConfiguration.Auth
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java:[97,20] cannot find symbol
  symbol:   method builder()
  location: class de.rwth.idsg.steve.SteveConfiguration.Ocpp
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/web/controller/AboutSettingsController.java:[70,45] cannot find symbol
  symbol:   method getSteveVersion()
  location: variable CONFIG of type de.rwth.idsg.steve.SteveConfiguration
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/web/controller/LogController.java:[57,13] cannot find symbol
  symbol:   variable log
  location: class de.rwth.idsg.steve.web.controller.LogController
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/service/MailService.java:[95,17] cannot find symbol
  symbol:   variable log
  location: class de.rwth.idsg.steve.service.MailService
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/service/MailService.java:[106,50] cannot find symbol
  symbol:   method getFrom()
  location: variable settings of type de.rwth.idsg.steve.repository.dto.MailSettings
[... 544 lines cut ...]
[ERROR]   location: variable p of type de.rwth.idsg.steve.repository.dto.InsertTransactionParams
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[232,13] cannot find symbol
[ERROR]   symbol:   variable log
[ERROR]   location: class de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl
[ERROR] /C:/Users/Tarik/Desktop/ocpp-steve/steve-steve-3.4.2/steve-steve-3.4.2/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[233,91] cannot find symbol
[ERROR]   symbol:   method getIdTag()
[ERROR]   location: variable p of type de.rwth.idsg.steve.repository.dto.InsertTransactionParams
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException`

---------------------------------------------------------------------------------------------------------------------------------
cmd:     `java -version`
output: java version "15"

cmd:     `javac -version`
output: javac 15

cmd: `mvn -version`
output: Apache Maven 3.6.3
---------------------------------------------------------------------------------------------------------------------------------

goekay, 2020-10-01:
it is possible that steve and the build tools we use are not ready for java 15 yet. please try with 13 or 11.

goekay, 2020-12-11:
we started using travis-ci with multiple build jobs, and java 15 is one of them. as you can see from here https://travis-ci.org/github/RWTH-i5-IDSG/steve/builds/749038022, it can build with openjdk15. therefore, i don't know what the issue is.

the compiler cannot find the source files related to the database schema. these should have been generated in a previous phase before the compile phase automatically. if you modified the maven build pipeline, this could be a problem.

goekay, 2021-03-22:
closing due to inactivity.

closed 2021-03-22 by goekay