steve-community/steve #410: Unable to retrieve meter values & transactions with SteVe using ocpp1.6 chargers

topgear490, 2020-10-01:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...3.4.3

Hello goekay and all. We have be using steve and all has worked well apart from one issue.

I am unable to get the meter values and transactions from the charging stations we have connected.

I am able to connect and use all the other ocpp 1.6j function without any problems.

has anyone come across this or able to help please?

thank you

goekay, 2020-10-01:
> I am unable to get the meter values and transactions from the charging stations we have connected.

what do you mean by that? in a normal operation, you do not get the meter values and transaction from the station. the station reports these to steve and steve stores them in the database. after that you read these from steve's database actually.

giorgio80, 2020-10-02:
Just to help. 

Ocpp interpretation of fields can , in real life, vary a lot between a producer and another one .
 If your station is sending metervalues it’s probably sending them into a non valid standard . 
Example it is sending watt as string “1000” instead as an int 1000 or a null value if not present. 

You should the. verify what your station is putting out from log and understand which values are not conform to standard. You should see an error in log 

From my perspective , I do firmwares , is totally compliant .

topgear490, 2020-10-02:
Thank you for the quick responses gentlemen.

So yes the transactions or meter values are not appearing on steve. (The stations are connected and making transactions/charging sessions.)

Do you know how I would retrieve them from the database or if I could get them to show on SteVe?

giorgio80, 2020-10-03:
are the metervalues sent in log?

goekay, 2020-10-06:
> The stations are connected and making transactions/charging sessions.

how do you know this? maybe station is making them but not reporting them to steve.

> Do you know how I would retrieve them from the database or if I could get them to show on SteVe?

the log tab of steve shows complete interaction (requests and responses) with stations. you should see the messages there. if you don't see anything happening, then station is indeed NOT reporting them.

goekay, 2020-12-11:
i am closing the issue due to inactivity.

closed 2020-12-11 by goekay