steve-community/steve #411: HTTPS to HTTP downgrade redirect on valid cert

Loop-Donsky, 2020-10-01:
Only filing as a bug because I'm not sure if this is intended or unintended behavior...

#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs // ==== We have diverged ever so slightly, but this is unrelated as I've looked through the issues and pull requests, and there's no work on 'https' or 'ssl' other than dep version bumping
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     :  3.4.2 ( SteveDB version 1.0.2 )
Operating system  : AWS ELasticBeanstalk EC2 Amazon Linux instance
JDK               : Corretto 11 running on 64bit Amazon Linux 2
Database      : Amazon RDS - MariaDB 10.4.8
```

####  Behavior

Steve is inside my VPC, behind a load balancer. The load balancer validates the SSL cert, and once it is, it forward the request to Steve over TCP. There's no way to get my hands on the original SSL cert distributed by AWS so I can't pass it to the Steve server during deploy, this is for security reasons, they don't allow a download/view. Because it's already inside the VPC verified by the SSL I would like to use Steve over the original connection but support the https protocol when it redirects... of course the browser shouldn't like that ( but it gets away with it ) and then I'm logged into an http session with my token... WAT? I'm hoping there's a way to maintain the protocol from the request regardless of Jetty's requirement for a cert to turn on HTTPS support.

#### Steps to Reproduce the Problem

  1. Spin up load balancer with SSL cert, expose port 443 on any given domain
  2. Spin up Steve in network with load balancer
  3. Point load balancer at Steve on http port 8080
  4. Try to login
  5. *choke* ( if you haven't exposed port 80 to the public on the load balancer, it should hang on a 302 failure )

#### Additional context

Any workaround suggestion is welcome as well

csamsel, 2020-10-04:
Why arent you just using a self signed certificate in Steve?

Loop-Donsky, 2020-10-05:
@csamsel I tried that so I could upload it. Assigned it to the ELB, but, if I use a self-signed cert, public CA's don't trust it, then neither do the browsers... Unless I'm misunderstanding you.

Loop-Donsky, 2020-10-05:
... wait do you mean use different certs, one at the ELB for public and one between the ELB and Steve? I'm not sure the ELB will happily use two certs for the same domain... but it's an idea...

csamsel, 2020-10-05:
Yes, that's what I meant, just a self signed certificate in Steve so it does not try to forward

Loop-Donsky, 2020-10-05:
Thank you sir, I will try this and then close the ticket when it's inevitably successful 👍

Loop-Donsky, 2020-10-07:
Still haven't had a chance to test this, I promise I'll close the ticket when I have 🍡

Loop-Donsky, 2021-02-05:
FYI: Starting this... I haven't forgotten, just haven't had time.

closed 2023-08-20 by goekay