steve-community/steve #419: Accessing ws://.../CentralSystemService/invalidId causes 500 error instead of 404

thg2k, 2020-10-15:
Opening a websocket to an non-existing chargeboxId causes an HTTP error 500 instead of something more reasonable like 404 to help troubleshooting the problem.

This is the body of the response:

```
HTTP ERROR 500
Problem accessing /steve/websocket/CentralSystemService/xtest1x. Reason:

    Server Error
Caused by:
org.springframework.web.util.NestedServletException: Request processing failed; nested exception is org.springframework.web.socket.server.HandshakeFailureException: ChargeBoxId 'xtest1x' is not recognized.
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1014)
	at org.springframework.web.servlet.FrameworkServlet.doGet(FrameworkServlet.java:898)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687)
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:876)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:542)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:146)
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:548)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerW
```

goekay, 2020-10-16:
spring's websocket implementation seems to wrap every other exception besides HandshakeFailureException into a HandshakeFailureException. so, even we throw another type of exception in this case, it will be converted into a 500. i get your point that it would be better. but at least you have the stack trace and a reasonable error message to troubleshoot.

thg2k, 2020-10-17:
> spring's websocket implementation seems to wrap every other exception besides HandshakeFailureException into a HandshakeFailureException

Do you think it would be possible to catch it somewhere? I guess the reason why it's not caught in the websocket code is because the websocket is not actually started yet, as we are still at HTTP level (i.e. the connection did not get upgraded).

It is a situation legally described in the websockets RFC, and I think the correct answer would be one of:

```
HTTP/ 404 Not Found
Content-Type: text/plain

Charge station not found.
```

Or, if we don't want to disclose information about the charge station ids:

```
HTTP/ 403 Forbidden
```

In the latter case, it would be not possible to distinguish between correct chargeboxId but wrong password, or incorrect chargeboxId.

Thx

goekay, 2020-10-17:
the relevant class for modification is this one: https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/OcppWebSocketUpgrader.java

as you see there, we throw HandshakeFailureException at three different places. HandshakeFailureException is the expected exception from the spring flow. our impl does not own the whole handshake and http/ws control pipeline. it rather is a step that fits into the general spring pipeline. therefore, we cannot do anything we desire.

i experimented with some different exception types. as i described above, these were wrapped into a HandshakeFailureException and 500. the user-facing error was still 500. of course, there would be a solution for this, but right not i don't have the capacity to look into it. feel free to experiment with the class above. if you stumple upon an improvement: we like PRs from community :)

goekay, 2022-03-05:
hey @thg2k, i know it's been a while and this is probably too late, but i did some improvements wrt. websocket connection establishment. and these enabled to address your issue as well. in particular....

so far we were using an extension of Jetty10RequestUpgradeStrategy that did some checks and allowed the upgrade to happen or not. this was a low-level interception and control mechanism with exits only via exceptions. i discovered that the proper place of doing these things should be HandshakeHandler where we can set the response codes, and early exit with ` return false` if we do not want to continue the process.

pls take a look at https://github.com/RWTH-i5-IDSG/steve/compare/issue419

goekay, 2022-03-07:
the referenced PR addresses the issue, and i am therefore closing it.

closed 2022-03-07 by goekay