steve-community/steve #428: SocketTimeoutException Exception while sending Remote Start Transaction

GRangaRao, 2020-10-27:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...3.4.3
Operating system  : ...Ubuntu 18.04
JDK               : ...JDK 11
Database          : ...MySQL 8.0
```

#### Expected Behavior
Sending RemoteStarttransaction to Charge Station should start the transaction 
 but fails with SocketTimeoutException
...

#### Actual Behavior
java.net.SocketTimeoutException: 60,000 milliseconds timeout on connection http-outgoing-39
...

#### Steps to Reproduce the Problem

  1. ...Charge Station is configured with Heartbeat interval 90 minutes
  2. ...heartbeat is seen as expected
  3. ...Remote Ocpp commands are failing to function

#### Additional context

...

goekay, 2020-11-12:
hey, sorry for the late reply.

is the problem only with RemoteStartTransaction operations? or, are other operations that you trigger from steve are failing as well?

if it is the RemoteStartTransaction only, i can imagine that the station internally tries to respect the command and start a transaction but fails for whatever reason. if this failure is not handled gracefully, the proper steps are not executed and no response will be sent to the caller (in this case you, the caller of RemoteStartTransaction). therefore, the http connection will wait for a response for 60 seconds and give up with this timeout.

if it is the latter, it must be a general connectivity or networking issue. but i doubt this possibility, because you wrote that heartbeats are working.

GRangaRao, 2020-11-13:
Thank you goekay for your response, have not seen often but only sometimes during tests

goekay, 2021-03-22:
closing due to inactivity.

closed 2021-03-22 by goekay