steve-community/steve #441: mariadb 10.4.16+: Malformed communication packet

barsnick, 2020-11-11:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.4.3
Operating system  : Linux (Fedora 32 x86_64)
JDK               : OpenJDK 11.0.9.11
Database          : MariaDB 10.4.x
```

#### Expected Behavior

```
[INFO ] 2020-11-10 12:15:30,864 org.hibernate.validator.internal.util.Version - HV000001: Hibernate Validator 6.1.4.Final
[INFO ] 2020-11-10 12:15:31,524 org.springframework.web.context.ContextLoader - Root WebApplicationContext initialized in 9181 ms
```

#### Actual Behavior

```
[INFO ] 2020-11-10 10:54:54,474 org.hibernate.validator.internal.util.Version - HV000001: Hibernate Validator 6.1.4.Final
[ERROR] 2020-11-10 10:54:55,594 org.springframework.web.context.ContextLoader - Context initialization failed
org.jooq.exception.DataAccessException: SQL [select timestampdiff(second, utc_timestamp(), current_timestamp()) from dual]; Malformed communication packet
        at org.jooq_3.12.4.MYSQL.debug(Unknown Source) ~[?:?]
        at org.jooq.impl.Tools.translate(Tools.java:2728) ~[jooq-3.12.4.jar:?]
        at org.jooq.impl.DefaultExecuteContext.sqlException(DefaultExecuteContext.java:755) ~[jooq-3.12.4.jar:?]
```

#### Steps to Reproduce the Problem

  1. update MariaDB from 10.4.14 to 10.4.16.
  2. Launch SteVe.

#### Additional context

A recent update of my Fedora 32 distribution updated MariaDB from 10.4.14(-2) to 10.4.16(-1). Instantly, database connections of the running instance broke, with the above quoted behavior (from a new launch, not the running instance).

Attached is a complete log from a launch attempt of SteVe, with full backtrace.

Reverting MariaDB to an older build fixed the issue.

There are other indications of this issue with MariaDB:
https://jira.mariadb.org/browse/MDEV-24121
https://jira.mariadb.org/browse/MDEV-24134

In the latter, [one comment](https://jira.mariadb.org/browse/MDEV-24134?focusedCommentId=171239&page=com.atlassian.jira.plugin.system.issuetabpanels:comment-tabpanel#comment-171239) says:

> Latest MariaDB releases have stricter packet validation as a security measure and these broken packets are now rejected.

(I couldn't find this in the MariaDB release notes though.)

I'm not sure which component in SteVe is at fault. Perhaps jooq (or whatever is responsible for creating the packet) needs a fix, but I didn't find any indication of this on their development page.

[steve.mariadb-10.4.16.log](https://github.com/RWTH-i5-IDSG/steve/files/5523041/steve.mariadb-10.4.16.log)

goekay, 2020-11-11:
thanks for the detailed report!

i became aware of this issue in [this PR](https://github.com/RWTH-i5-IDSG/steve/pull/434). however, the issue is independent of the version bump in there. the automated builds were triggered after the update and therefore most recent dependencies were pulled which included newest mariadb.

the devs of the mariadb stated that 1) the database drivers (jdbc) are at fault here (therefore: a new version of mysql jdbc might fix it)  and 2) they will relax the validation with an upcoming mariadb version. 

we have a github dependency plugin that tracks the version of dependencies. therefore, we can catch the first point easily. other than that i don't think there is something that steve can do... unfortunately.

barsnick, 2020-11-12:
Ah, nice, I couldn't figure out that you are aware of the issue.

I would have "pinned" my mariadb installation and waited for an update of MySQL Connector/J (thanks for pointing out the component), but it turns out, MariaDB devs have changed their minds and hastily released an update:
https://mariadb.org/mariadb-10-5-8-10-4-17-10-3-27-and-10-2-36-now-available/

Fedora has a new release of the mariadb package (which hasn't quite arrived in F32 repos yet though). I assume I'll get the update soon. And all others can also breathe a sigh of relief.

goekay, 2020-11-12:
:) i am closing this issue, since we sorted this out. i hope this is okay for you.

closed 2020-11-12 by goekay