steve-community/steve #443: FTP files are not downloading

rdc-Green, 2020-11-12:
SteVe Version     : 3.4.1
Operating system  : Ubuntu
JDK               : ...
Database          : mySQL

I have created an FTP link to my server using port 21. I have then uploaded the latest firmware version supplied by Alfen to the server using Filezilla.
I then execute the command in the SteVe UI by copying the file URL from Filezilla into the location in 'Update Firmware'

The task is completed in SteVe, but it says download failed in the log, with no other explanation

[INFO ] 2020-11-12 08:09:26,546 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=NGV_0063, sessionId=a86e4dfe-8c4b-3fc4-0c77-18c94c08054a] Received: [3,"b158c85d-a2a6-42f7-9c6a-775dec630fc7",{}]
[INFO ] 2020-11-12 08:09:26,661 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=NGV_0063, sessionId=a86e4dfe-8c4b-3fc4-0c77-18c94c08054a] Received: [2,"7","FirmwareStatusNotification",{"status":"Downloading"}]
[INFO ] 2020-11-12 08:09:26,687 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=NGV_0063, sessionId=a86e4dfe-8c4b-3fc4-0c77-18c94c08054a] Sending: [3,"7",{}]
[INFO ] 2020-11-12 08:09:26,727 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=NGV_0063, sessionId=a86e4dfe-8c4b-3fc4-0c77-18c94c08054a] Received: [2,"8","FirmwareStatusNotification",{"status":"DownloadFailed"}]

Am I missing a step?

Thank you

goekay, 2020-11-12:
the following are two separate processes:
1. at steve: sending a message to the station and communicating a link about the location of the firmware file.
2. at station: based on the received link, the station tries to download the file.

the logs indicate that the first step is successful. at the start of 2, the station sends a message it starts the download, but then reports that it failed. this also means, that steve is not at fault here. it did what it could. the station failed during the download process.

unfortunately, there are no details about why it failed. the firmware link might be incorrect. maybe the station cannot access the ftp server, or the link, or your ftp server requires authentication, or or or. it sounds like a networking and rights management problem.

rdc-Green, 2020-11-12:
Thank you @gregorschatz I appreciate you taking the time to respond. I'll test other options

closed 2020-11-12 by rdc-Green