steve-community/steve #444: question: Dockerfile for Steve only

lblabr, 2020-11-13:
i have an mariadb server running and would like to persist the information there. so i tried to adopt your dockerfile to an steve only... but i can't get it work

```
FROM maven:3.6.1-jdk-11

ENV LANG=C.UTF-8 LC_ALL=C.UTF-8

ENV DB_HOST=10.100.10.201
ENV DB_PORT=3308
ENV DB_USERNAME=u_steve
ENV DB_PASSWORD=changeme
ENV DB_DATABASE=steve

MAINTAINER Ling Li

EXPOSE 8180
EXPOSE 8443

WORKDIR /code

ADD https://github.com/RWTH-i5-IDSG/steve/archive/steve-3.4.3.tar.gz /tmp/code/ 
RUN tar -zxf /tmp/code/steve-3.4.3.tar.gz -C /code steve-steve-3.4.3/ --strip-components 1

RUN sed -i 's|${db.ip}|${env.DB_HOST}|g' pom.xml
RUN sed -i 's|${db.port}|${env.DB_PORT}|g' pom.xml
RUN sed -i 's|${db.user}|${env.DB_USERNAME}|g' pom.xml
RUN sed -i 's|${db.password}|${env.DB_PASSWORD}|g' pom.xml
RUN sed -i 's|${db.schema}|${env.DB_DATABASE}|g' pom.xml

# Wait for the db to startup(via dockerize), then 
# Build and run steve, requires a db to be available on port 3306
CMD mvn clean package -Pdocker -Djdk.tls.client.protocols="TLSv1,TLSv1.1,TLSv1.2" && \
	java -jar target/steve.jar

```

i don't know how to fix:

```
,[INFO] BUILD SUCCESS
,[INFO] ------------------------------------------------------------------------
,[INFO] Total time:  06:15 min
,[INFO] Finished at: 2020-11-13T20:40:08Z
,[INFO] ------------------------------------------------------------------------
,[INFO ] 2020-11-13 20:40:22,961 de.rwth.idsg.steve.utils.PropertiesFileLoader (main) - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
,[INFO ] 2020-11-13 20:40:23,905 de.rwth.idsg.steve.Application (main) - Loaded the properties. Starting with the 'PROD' profile
,[INFO ] 2020-11-13 20:40:24,476 de.rwth.idsg.steve.Application (main) - Date/time zone of the application is set to UTC. Current date/time: 2020-11-13T20:40:24.206Z
,Log file: Not available
,Starting.[INFO ] 2020-11-13 20:40:25,594 org.eclipse.jetty.util.log (main) - Logging initialized @16154ms to org.eclipse.jetty.util.log.Slf4jLog
,....[INFO ] 2020-11-13 20:40:28,170 org.eclipse.jetty.server.Server (main) - jetty-9.4.19.v20190610; built: 2019-06-10T16:30:51.723Z; git: afcf563148970e98786327af5e07c261fda175d3; jvm 11.0.4+11
,...[INFO ] 2020-11-13 20:40:30,130 org.eclipse.jetty.server.session (main) - DefaultSessionIdManager workerName=node0
,[INFO ] 2020-11-13 20:40:30,130 org.eclipse.jetty.server.session (main) - No SessionScavenger set, using defaults
,[INFO ] 2020-11-13 20:40:30,135 org.eclipse.jetty.server.session (main) - node0 Scavenging every 600000ms
,.[INFO ] 2020-11-13 20:40:30,237 org.eclipse.jetty.server.handler.ContextHandler.steve (main) - Initializing Spring root WebApplicationContext
,[INFO ] 2020-11-13 20:40:30,239 org.springframework.web.context.ContextLoader (main) - Root WebApplicationContext: initialization started
,.....................[INFO ] 2020-11-13 20:40:43,413 com.zaxxer.hikari.HikariDataSource (main) - HikariPool-1 - Starting...
,..... FAILED!
,Please refer to the log file for details
,[ERROR] 2020-11-13 20:40:45,925 com.zaxxer.hikari.pool.HikariPool (main) - HikariPool-1 - Exception during pool initialization.
,com.mysql.cj.jdbc.exceptions.CommunicationsException: Communications link failure
,
,The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
,	at com.mysql.cj.jdbc.exceptions.SQLError.createCommunicationsException(SQLError.java:174) ~[mysql-connector-java-8.0.19.jar:8.0.19]

```

any help would be appreciated!

Thanks in advance

Lars

goekay, 2020-11-16:
seems like the same issue as in https://github.com/RWTH-i5-IDSG/steve/issues/441

lblabr, 2020-11-18:
for the DB Docker is used the same image like you in the docker-compose.yml... The built with your docker compose is running, having to separate docker container does not work

tvand, 2022-11-07:
I use the following Dockerfile that works for me:
```
FROM amazoncorretto:11.0.17-alpine3.16

ENV LANG=C.UTF-8 LC_ALL=C.UTF-8

EXPOSE 8180
WORKDIR /code

# Copy the application's code
COPY target/steve.jar /code
COPY target/libs /code/libs
COPY src/main/resources/config/prod/main.properties /code/steve.properties

CMD java -Dmain.properties=/code/steve.properties -jar steve.jar
```
The separate properties file allows an external file to be mounted into the container, so that no credentials are being stored in the JAR and the container.

Moreover I extended the pom.xml to employ an optional local properties file for the build:
```
            <!-- Read from main.properties the DB configuration -->
            <plugin>
                <groupId>org.codehaus.mojo</groupId>
                <artifactId>properties-maven-plugin</artifactId>
                <version>1.1.0</version>
                <executions>
                    <execution>
                        <phase>initialize</phase>
                        <goals>
                            <goal>read-project-properties</goal>
                        </goals>
                        <configuration>
                            <files>
                                <file>src/main/resources/config/${envName}/main.properties</file>
                                <file>steve.properties</file>
                            </files>
                            <quiet>true</quiet>
                        </configuration>
                    </execution>
                </executions>
            </plugin>
```

closed 2023-08-20 by goekay