steve-community/steve #446: FTP server not downloading or uploading

rdc-Green, 2020-11-17:
I am trying to use Filezilla to upload files to Alfen charge points. SteVe is hosted on Azure. I can connect in Filezilla and FTPtest.net returns a successful connection, but I still get Download failed in the SteVe log. I also tried Get Configuration and get Upload failed too.

If you have had successful FTP transfers, could you please suggest an FTP server to use. If you have used Filezilla, was there anything specific that you had to do in order to communicate with SteVe?

Many thanks

goekay, 2020-11-17:
> If you have had successful FTP transfers, could you please suggest an FTP server to use. If you have used Filezilla, was there anything specific that you had to do in order to communicate with SteVe?

i am not in the business of charging stations. therefore, i don't have any charging stations in production and cannot speak from experience.

> I also tried Get Configuration and get Upload failed too.

can you expand on this? what failed exactly? based on the other issue you opened (https://github.com/RWTH-i5-IDSG/steve/issues/443), i thought that the communication is happening in both directions (steve->station and station->steve).

please be mindful when opening issues and asking questions: github issues are not a customer support service. we do not and cannot provide such a service. the issue has to be about a problem in steve. but so far, it does not look like that.

rdc-Green, 2020-11-17:
Hi @goekay thank you again for responding. I understand its not a customer support service, but as far as i can tell the FTP server is operating correctly based on my test using ftptest.net and a successful connection in Filezilla. I am trying to work out if it is a steve problem. I can only see what's in the log, which is just a 'failed' response. I'm hoping that somebody who has had a successful transfer can help me understand how they did that.

I tried the Get Configuration command in steve with the file location being a file I have set up in Filezilla. This also fails to upload and I can only see a 'Failed Upload' response in the log. It looks like the communication is working in that it sends a failed response but I don't know why it has failed.

I'm also working on the Filezilla side to see if there is something there blocking the uploading and downloading of files.

goekay, 2020-11-17:
> I tried the Get Configuration command in steve with the file location being a file I have set up in Filezilla.

i have trouble following you. with "get configuration" command, you only select the name of an operation for which you want the station send you more info. what do you mean with "with the file location being a file I have set up in Filezilla."? you do not specify a location with "get configuration". therefore i cannot understand this either:

> This also fails to upload and I can only see a 'Failed Upload' response in the log.

> I am trying to work out if it is a steve problem. 

of course i cannot fully exclude the probability of an issue with steve. but the logs you provided in the other issue suggest otherwise.  with the "upload firmware" operation, steve just sends the link of the file. this operation only communicates where to find the firmware. if this succeeds, after that, it is no more a concern of steve. if possible, you should investigate the logs of the station.

also keep in mind, it can also happen that the station expects the file to be at the same host where steve is. if steve is hosted at X but the file is at Y the station has problems accessing Y.

i am leaving this issue open for visibility. maybe others can chime in. i don't think i can be helpful furthermore.

rdc-Green, 2020-11-17:
sorry @goekay i meant to say Get Diagnostics not Get Configuration. I will check the station log too, thats a good idea that I hadn't considered.

steve is hosted in the same location as the FTP server on Azure. 

Thank you for leaving it open for the time being. Hopefully somebody else has resolved this problem already. I will update if I find a solution that hopefully shows it isn't to do with steve!

rdc-Green, 2020-11-19:
I have solved my issue. You were right @goekay it wasn't a SteVe problem!! When using ftp with Alfen charge points you need to include passwd in the URL and you cannot use TLS. Thank you for your support with this and pointing me in the direction of the charger log rather than the SteVe log!

rdc-Green, 2023-11-09:
re-opening this as I have spotted an issue with ftp requests.

When trying to 'Get Diagnostics' i use my public ip address in the url. This works for most situations, but for some reason when the request is processed by the charge point it tries to connect to the internal server ip address. Mostly it then switches to trying to communicate with the public ip address and it works.

Does anybody know if this is related to the SteVe ftp process (requesting public and internal ip connections)?

Many thanks